prusa3d/PrusaSlicer #1176: Custom Printer Profil Window size

Medalknight, 2018-08-31:
### Version
1.41.0+

### Operating system type + version
MacOS 10.13.6

### Behavior
* _Describe the problem_
My own custom profile for the printer resize the quick select window from slic3r.
* _Steps needed to reproduce the problem_
generate and use a custom profil 
* _Expected Results_
no resize of the windows
* _Actual Results_
window gets bigger
  * _Screenshots from __*Slic3r*__ preview are preferred_
[image]

[image]

_Is this a new feature request?_
No

bubnikv, 2018-09-27:
@YuSanka please make the size of the right panel fixed.

bubnikv, 2019-04-13:
Right panel size is fixed in https://github.com/prusa3d/Slic3r/releases/tag/version_1.42.0-beta1

closed 2019-04-13 by bubnikv