prusa3d/PrusaSlicer #1177: Please post the binaries, especially for linux users

vertigo235, 2018-08-31:
The https://www.prusa3d.com/drivers/ page directs linux users here for releases, but there is no 1.41 final release binaries. 

### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

domesticatedviking, 2018-08-31:
Yes, please post the binaries and appimage build!

bubnikv, 2018-09-01:
I will post them today. Closing for now.

closed 2018-09-01 by bubnikv