prusa3d/PrusaSlicer #1178: Recent configuration bundle push broke configuration

kortschak, 2018-09-01:
### Version
Previously Slic3rPE-1.41.0-alpha2+linux64-full-201807211541.AppImage but now Slic3rPE-1.41.0-beta+linux64-full-201808061625.AppImage

### Operating system type + version
Linux Ubuntu 16.04.4

### Behavior
A configuration bundle was suggested while running Slic3rPE-1.41.0-alpha2 and allowed. After closing and re-starting another session of Slic3r a dialogue complained that the version of Slic3r is not compatible with the installed configuration bundles. No version change had been made and no earlier version of Slic3r had been run.

So try upgrading to 1.41.0-beta and see exactly the same problem. Though with the Kafkaesque complaint that the minimum version for the configuration bundle is 1.41.0; the latest available release is [1.41.0-beta](https://github.com/prusa3d/Slic3r/releases/tag/version_1.41.0-beta).

[image]

bubnikv, 2018-09-01:
Slic3r reads
https://github.com/prusa3d/Slic3r-settings/blob/master/live/PrusaResearch/index.idx
where it takes into account the first min_slic3r_version seen from the top. I have adjusted the minimum slic3r version, so after the next update you should be fine. Sorry for the inconvenience.

bubnikv, 2018-09-01:
Is it working for you now?