prusa3d/PrusaSlicer #1179: Camera issue

tkircher, 2018-09-01:
### Version
1.41.0-beta-1

### Operating system type + version
MacOS Sierra

### Behavior
This behavior isn't specific to the included STL, but it's provided for reference. Any large object will do this. Import, then scale to fit on the print bed. Then any attempt to change the camera perspective causes everything to disappear. Probably due to clipping. The correct behavior would be to have the print plate centered, then zooming in would zoom centered on the plate, as intuitively expected. The biggest problem is that there's no easy way to get the system centered again. There should be some feature like in other programs where you can reset the view, and it defaults the camera to a fixed distance away from the print bed, pointed at the center.

#### STL/Config (.ZIP) where problem occurs

[Gort_large.stl.zip](https://github.com/prusa3d/Slic3r/files/2342098/Gort_large.stl.zip)

austinvojta, 2018-09-05:
I miss when the "0" key would reset the camera to a fixed distance away along with resetting to an isometric view. Not sure why they removed that in a recent update, but I too would like that back.

FidelCapo, 2018-09-05:
Thank you for report. I can reproduce this issue.

bubnikv, 2018-09-27:
There are (undocumented) hot keys "z" and "b" to zoom to active object or to the print bed.
We should document these hot keys @YuSanka please would you add these to the menu?
I know this is not a solution, but a workaround.

We have to think about the concept of the center of rotation. The issue is more complex for the ortographic camera than it seems, so we will do a proper fix first in the next major release.

bubnikv, 2018-12-12:
In addition to the "z" and "b", we now keep the camera focus inside the bounding box of the scene, so you will never lose focus. This will be part of the upcoming Slic3r PE 1.42 alpha relese.
Closing.

closed 2018-12-12 by bubnikv