prusa3d/PrusaSlicer #1180: Slic3r installs incompatible configuration

ctag, 2018-09-01:
### Version
(About menu is disabled) Package is version: [1.40.1-2](https://aur.archlinux.org/packages/slic3r-prusa3d)

### Operating system type + version
Archlinux

### Behavior
Slic3r prompted me to update the profiles with new MM updates, I accepted. Now Slic3r does not start, apparently because the profiles are incompatible, and the distro package has not yet updated.

I expect Slic3r to not install profiles which require a higher version of Slic3r than is currently running.

Screenshot: https://i.imgur.com/mWWRzZk.png

bubnikv, 2018-09-01:
Unfortunately it happened, that I have pushed a wrong configuration index
online. The index online should be fine now, so your Slic3r configuration
should be fixed after you let slic3r to reconfigure. In worst case you may
need to clean your Slic3rPE config directory. Or better update your Slic3r
to 1.41.0 final.

ctag, 2018-09-01:
Thanks for the clarification.

It looks like the updated version won't be available for my distro for a while, and I wasn't able to build it manually. So I reconfigured 1.40.1 and it appears to be working properly again.

closed 2018-09-01 by ctag