prusa3d/PrusaSlicer #1183: Auto-Center not working

FlailAway, 2018-09-01:
### Version
Slic3r Prusa edition-1.41.0+Linux64
Downloaded the .Appimage this morning

### Operating system type + version
Ubuntu 16.04 all up to date

### Behavior
Subject says it all. For a Cartesian printer, it is placing the incoming object at 0,0 (Left-Front corner) rather than the center of the bed. I have to manually move the object before slicing. The Beta version was fine for centering.

Is there a setting for this? I could not find anything.

FidelCapo, 2018-12-13:
Can you please export and send me config.ini (File-export config..). Thank you.

rtyr, 2020-03-29:
No response. Auto-center seems fine in the current PrusaSlicer 2.2.0. Closing.

closed 2020-03-29 by rtyr