prusa3d/PrusaSlicer #1185: Bridge infill doesn't cross perimeters below

ScanTraxx27, 2018-09-02:
### Version
Slic3r PE 1.41.0+win64

### Operating system type + version
Windows 10 x64 - Version 10.0.17134

### Behavior
* Describe the problem:
Bridging can only be effective if the bridge lines cross the perimeters lying below so that the beginning and end of every bridge line have something to rest on. Otherwise the lines will just sack down without any halt.
However, this is exactly what Slic3r does. Thus, bridging is really bad and not at all fulfilling it's purpose.

* Steps needed to reproduce the problem:
1. Import any STL with bridging features into Slic3r.
2. (optional) Enable "detect bridging perimeters". (The issue also occurs with this option disabled)

* Expected Results:
Bridge infill lines cross perimeters below. Here's an example from Cura:
[image]

* Actual Results:
Bridge infill lines DON'T cross perimeters below.
[image]

#### STL/Config (.ZIP) where problem occurs
[STL+Config.zip](https://github.com/prusa3d/Slic3r/files/2343309/STL%2BConfig.zip)

supermerill, 2018-09-04:
[image]

can't reproduce

ScanTraxx27, 2018-09-04:
Thanks for the reply.
I've just tried it once again and suddenly it worked - I got the same result as you!
Then I realized that last time I had infill set to 0% because there's no need for infill here (except maybe for extending the layer time) and that reproduced the problem.

Soo, should I just let this issue be closed and open a new one with the correct steps?

supermerill, 2018-09-04:
just rename this issue, i will be able to create a fix now i understand the problem.

ScanTraxx27, 2018-09-04:
Thanks! Good luck on fixing the problem!

bubnikv, 2018-09-27:
This is a similar issue connected to no infill.
https://github.com/prusa3d/Slic3r/issues/442

supermerill, 2018-10-03:
@ScanTraxx27 you can test the fix in my fork.
@bubnikv my fix is in the commit linked here. Do you want a pr?

Gebhartj, 2019-07-28:
To solve this problem I use a 1% concentric infill

bubnikv, 2019-09-09:
Fixed with 48ecbe7
The fix will likely be part of PrusaSlicer 2.2.

Closing.

closed 2019-09-09 by bubnikv

Parker-Drouillard, 2020-04-14:
I see this was closed and it was stated that it would be fixed in V2.2, It is quite some time later and we are now on 2.2 and yet I am still experiencing this issue of the bridges not even attempting to anchor

bubnikv, 2020-04-14:
@ScanTraxx27 Provide data. Different issues manifest with similar effects.