prusa3d/PrusaSlicer #1188: Single Extruder Multi Material Option injects M907 into Gcode

GrkMunga, 2018-09-02:
### Version
Slic3rPE-1.41.0+win64

### Operating system type + version
Windows 10

### Behavior
Enabling the Single Nozzle Multi Material option in Printer Settings causes the M907 command to be inserted into the gcode. This causes the Wifi Duet to enter an error state and stop the print. Oddly tho it the M907 command is only inserted when the option is enabled. When disabled, that command is nowhere to be seen in the gcode. Attached is an example of the gcode but with the option enabled and disabled.

Enabling the Single Nozzle Multi Material setting is the only way to create a purge block as without it, it will never appear.

[Gcode.zip](https://github.com/prusa3d/Slic3r/files/2343757/Gcode.zip)

bubnikv, 2018-09-27:
M907 is to control the extruder motor current, which we use to increase extruder motor current during the filament exchange to overcome expansion at the tip of the filament when pulled from the melt zone. Sorry, there is currently no way to disable this code, it is hard coded.

GrkMunga, 2018-09-28:
Ahh I see. Will there be plans in the future to disable this code for other printers or any other workaround? Currently the only way to get a purge block is to have Single Nozzle Multi Material enabled, but this causes issues.

lukasmatena, 2019-06-07:
I believe this has been solved long ago by https://github.com/prusa3d/PrusaSlicer/commit/6da83c797690afbb4c455e2a90e54fe3d6cc9ed3, which made the increase of current for ramming optional. Closing.

closed 2019-06-07 by lukasmatena