prusa3d/PrusaSlicer #1189: Going to settings crash(freeze) the whole slic3r

alomex21, 2018-09-03:
### Version
1.41.0-beta+win64

### Operating system type + version
Windows 10 Pro 64-bit
### Behavior
When importing any model, trying to open setting on a specific model, it crash the application without creating the 'Not responding' message from windows
Cleaned every file created from the zip extracted and the profiles, but still freeze.

bubnikv, 2018-09-03:
>  When importing any model, trying to open setting on a specific model, it
crash the application without creating the 'Not responding' message from
windows

What do you mean with "opening settings"? Double clicking on an object?

The behavior described looks similar to
https://github.com/prusa3d/Slic3r/issues/1123

Would you maybe work with @vojtechkral to install a debugger and produce us
a crash report, as described at the end of #1123? Thanks.

bubnikv, 2018-09-03:
Does the slic3r crash, or does it "just" close? Are you technical enough to
enable crash dump collection for Slic3r and provide us with the crash dumps?

https://docs.microsoft.com/en-us/windows/desktop/wer/collecting-user-mode-dumps

bubnikv, 2018-09-03:
@vojtechkral created a short tutorial on how to produce a crashdump: https://github.com/prusa3d/Slic3r/wiki/Enabling-crashdump-collection-on-Windows

Would you please be so kind as to create the crashdump and share it with us so that we can have a look at what's going on in there? Thanks!

alomex21, 2018-09-04:
Hey I'm back, sorry for the delay.

Ok I tried the crashdump but sadly the application it just "freezes", hence it doesn't produce any report for it. Double clicking a model also freeze the slic3r aside of the right click on model->Settings. I've tried with the newest release(1.41.0) and also having the same problem. 
Is there anything that Slic3r creates aside of the application folder itself?

alomex21, 2018-09-05:
@bubnikv  I found the problem! I installed the Prusa drivers(https://www.prusa3d.com/drivers/) for the mk2s and while replacing files from GitHub from new versions time to time, it create some weird problems that I can’t describe precisely. What I did for fixing the problem is put the folder from the zip of a latest version of slic3r to the desktop or anywhere except the Prusa3D folder from default.

alomex21, 2018-09-05:
I Cleaned the cache from AppData and now it is working from Prusa3D folder

closed 2018-09-05 by alomex21