prusa3d/PrusaSlicer #1192: Slic3r will not export if you delete an object off the bed

jweaver100, 2018-09-03:
### Version
1.41.0+win64

### Operating system type + version
Windows 7

### Behavior

In Slic3r 1.40.0, if you had items off the bed (blue) you could not slicer or export.. But in 1.41.0 its now possible to slice when you move objects off the bed (Which is good).

But, if you delete an object whilst exporting, you cannot slice/export any more jobs until you restart slicer.

To reproduce

1. Add 2 items
2. Move one to the bed (green/yellow)
3. Move one OFF the bed (Blue)
4. Export
5. Whilst exporting gcode is writing the file delete the blue object
6. Now move the green/yellow obeject (or add something else) and try exporting again and you get the error "Another export job is currently running".

After this, you cannot export any further jobs and you have to restart slicer to fix it.

bubnikv, 2018-09-04:
This looks like one of the "background processing" issues, which we hope to fix by refactoring the background processing from Perl to C++. We hope to do it for the 1.42 release, keep your fingers crossed.

bubnikv, 2018-12-11:
The code responsible for your issue was rewritten from Perl to C++, also the background processing should now work reliably. Please retest with 1.42 alpha once it is released this or the next week. Closing.

closed 2018-12-11 by bubnikv