prusa3d/PrusaSlicer #1193: Auto-center parts conflicts with new Arrange parts

Area5142, 2018-09-04:
### Version
Version 1.41.0+win64

### Operating system type + version
Windows 10 Pro x64

### Behavior
Using _Arrange_ to arrange 12 copies of a simple part places 11 parts on the build plate and 1 on the virtual build plate.
Turning on _Auto-center parts_ in _Preferences_ will center both the physical build plate and the virtual as one. The arranged parts are moved off the build plate and made unprintable (see screen copies).

#### STL/Config (.ZIP) where problem occurs
Used test shape:
[Key Test Shape.zip](https://github.com/prusa3d/Slic3r/files/2349059/Key.Test.Shape.zip)

_Arrange_ 12 copies of test shape with _Auto-center parts_ unchecked (off):
[image]

_Arrange_ 12 copies of test shape with _Auto-center parts_ checked (on):
[image]

Solution: _Auto-center parts_ should only work on individual build plates (Physical and virtual) to center parts.

tamasmeszaros, 2018-09-11:
Hello and thank you for the feedback. The auto-center is a known bug, it will be addressed in the next releases. As to the arrangement: this is a very interesting case where there seems to be space left for the last item but in reality, if you consider the minimum distance which has to be maintained, there is no room left. If you enlarge the print bed slightly (6 mm is enough for the X axis) then the configuration will fit perfectly.

It may happen that there is in fact space for the last item but the solver cannot find it properly due to numerical or accuracy issues. We have a few test cases for similar configurations which failed similarly and we fine tuned the accuracy/performance for the algorithm. If there will be more issues like this where there is an evident fit for the last item but it ends up on the virtual bed, we can still exchange some accuracy for a slight performance cost.

Best regards,
Tamás

tamasmeszaros, 2018-09-13:
I'm sorry but at the moment it is hard coded into the application and you cannot change it. The philosophy is that the software should be able to tell the minimum distance for safe printing so you don't have to think about it.

Thank you for the compliment. This is only the first version of the auto arrange and it should get much better with support for concavities and part-in-part placement. Unfortunately, it's unclear which release will include these upgrades.

Best regards,
Tamás

rtyr, 2021-01-21:
I believe this is already fixed in the current PrusaSlicer 2.3.0. Minimum distance between the objects can also be configured now.
Closing.

closed 2021-01-21 by rtyr