prusa3d/PrusaSlicer #1194: Application hangs indefinitely during repair of STL model

M-a-t-t-S-m-i-t-h, 2018-09-05:
### Version
1.41.0
SHA1 ID: 22569de

### Operating system type + version
Windows 10 64 bit

### Perl + wxWidgets binary distribution
wperl64d-5.24.0-2018-03-02.7z

### Additional dependencies
slic3r-destdir-64.7z

### Build instruction
cmake .. -G "Visual Studio 12 Win64" -DCMAKE_CONFIGURATION_TYPES=RelWithDebInfo

### Debugger
Microsoft Visual Studio Community 2013
Version 12.0.40629.00 Update 5

### Behavior
The application hangs indefinitely while attempting to repair a STL model. The dialog "Processing input file" will be displayed, but with no advancement of the progress bar. 

### Knowledge so far
After some initial looking into of the problem, I believe the bug is in the ADMesh library, or in Slic3r's utilization of it. The particular model attached is detected as having 1200 facets with one bad edge. Performing two iterations of the stl_check_facets_nearby repair will fix 1160 of them (Incidentally, running a third iteration with its larger tolerance will fix the remaining 40 facets and allow the repair to proceed successfully. Unfortunately, the ability to increase this parameter neither is available to the end-user, nor does it address the underlying bug). The next stage stl_remove_unconnected_facets removes 764 degenerate facets (and prints a great deal of warnings to the console. See console log attached). Finally, stl_fill_holes is called and runs into a seemingly infinite loop.

As of right now, I don't know whether stl_fill_holes is solely at fault, or whether stl_check_facets_nearby or stl_remove_unconnected_facets left the stl object in a bad/corrupted state.

### Final thoughts
The repair function stl_remove_unconnected_facets presented a large number of faux-asserts in the way of "this is wrong" statements to the console. I don't know enough about this function to say whether this message indicates a situation that can eventually resolve itself, or whether it's an early diagnosis of the latter problem I'm experiencing. If it's the second one, I'd recommend that in addition to printing the message to the screen, that stl->error be set, the function returned from, and the user prompted of failure-to-import. 

### STL/Config (.ZIP) where problem occurs
Model: Frame_body_no_PEG.STL
Website: https://www.youmagine.com/designs/fallout-4-mini-nuke-redicubricks

[Frame_body_no_PEG.zip](https://github.com/prusa3d/Slic3r/files/2350605/Frame_body_no_PEG.zip)

### Console Log
I have attached a console log produced during the attempted import of this STL model. During this run I had enabled a few additional application print-outs:

- Enabled #if 1 
C:\git\Prusa3D\Slic3r\xs\src\admesh\connect.cpp:120

- Uncommented two printf's and corresponding fix count
C:\git\Prusa3D\Slic3r\xs\src\libslic3r\TriangleMesh.cpp:182, 189, 191

[console.txt](https://github.com/prusa3d/Slic3r/files/2350721/console.txt)

bubnikv, 2018-09-27:
Thanks for the very helpful investigation. The admesh library is in a need of a lot of love indeed.

bubnikv, 2018-10-05:
Please test the vb_slicing_fix2 branch. 

I have disabled the hole filling completely, and I am now connecting the remaining open contours into loops over maximum 2mm wide gaps during the actual slicing. This approach seems to be much more robust and IMHO this is what S3D or Cura do.

There is not a single good heuristics to deal with broken models. I think we will let this fix go into the wild, and we will wait for the response.

closed 2018-10-05 by bubnikv