prusa3d/PrusaSlicer #1200: MM V1 Issue

Calvin6, 2018-09-05:
### Version 1.41
_Version of Slic3r used goes here_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version Windows 10 64Bits

### Behavior
* _Describe the problem_
I have a prusa Mk2s with multimaterial v1, and the gcode generated by the slicer gets the message "incomplete file" in every configuration I selected, then I run the print anyway and it turns out to be correct so I think it´s maybe a compatibility problem between the multimaterial v2 and v1 presets code.
* _Steps needed to reproduce the problem_
Add a multi-color part, assign the colors and slice the code with wenever configuration you want, then export the gcode to the SD. When the printer read the file shows the message "incomplete file".
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
The printer should read the code just fine and use it
* _Actual Results_
The printer read the code as "incomplete file"
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_ NO

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2018-09-06:
This is really a request to the firmware team. The firmware checks for the
"motor power off" M code at the end of the G-code file, but it there are
parameters stored there as comments. Now we have more parameters stored by
Slic3r, so the firmware guys should increase the range from the end of the
file to look for the "motor power off" M code. They did change that for the
MK3 and MK2.5 firmware, but not for the MK2 / MK2MM firmware yet.

Closing in the Slic3r PE issue tracker.

closed 2018-09-06 by bubnikv

Calvin6, 2018-09-07:
Thanks for the explanation!