prusa3d/PrusaSlicer #1201: 2D bed doesn't display

mikezs, 2018-09-06:
### Version
1.41.0+

### Operating system type + version
macOS 10.13.6

### Behavior
* 2D bed preview disappeared in alpha3 of 1.41.0:
[image]
[image]

And is still not present in the very latest version:
[image]

This is obviously deliberate, but there's no commit message or comment to say why, and wasn't in the changelog. I just find a commit here: https://github.com/prusa3d/Slic3r/commit/4243c7d84aff757108d365c147955210c7058b86 by @enricoturri1966

bubnikv, 2018-09-06:
@enricoturri1966 Would you please look into it? Thanks.

bubnikv, 2018-09-06:
> say why, and wasn't in the changelog. I just find a commit here: 4243c7d

> 2D bed preview disappeared in alpha3 of 1.41.0:

It is not a bug, it's a feature. We removed it on purpose, as not many were
using it.

You may set the top view using the View menu to achieve very much the same
effect. Does it satisfy you?

mikezs, 2018-09-06:
> not many were using it.

I was under the impression it was fairly well used, as it's one of the top level screen elements. I've certainly be using it for a number of years (before PE). Is there any technical issues for keeping it? I don't see why it not being heavily used by everyone is a reason to remove?

> You may set the top view using the View menu to achieve very much the same effect. Does it satisfy you?

It's similar, and also very useful, but having it in 2D view previews the skirt and it's much easier to position things. If there's no technical limitation to having it I think a number of people would like it re-instated :)

bubnikv, 2018-09-27:
> Is there any technical issues for keeping it?

We are getting rid of Perl completely, and we are also in the process of cleaning up the user interface. Any code that we may remove is something we do not need to maintain anymore.

The 2D view is really a legacy of Slic3r before the 3D view has been integrated, and to support very low end or old computers. We will replace the 2D view most likely by a camera view gizmo (look from the top). Actually you can already do that with the "1" or "2" hot keys or through the "View" menu.

Closing for now, as we are revamping the UI anyway, so everything will be different and we will get a new set of issues.

closed 2018-09-27 by bubnikv

mikezs, 2018-09-27:
There's a few nice features that the 2D view has, when used with the 3D view:
- Always zoomed to fill screen
- Shows skirt around parts
- Not rotated by accident if 
- Becomes a second view for the bed

I'm a professional programmer (although not C++ or Perl) so I'll have a go at implementing the features I want after the interface is revamped. Happy to leave closed.

bubnikv, 2018-09-27:
Thanks for your feedback.

gsedej, 2018-10-22:
> We are getting rid of Perl completely, and we are also in the process of cleaning up the user interface. Any code that we may remove is something we do not need to maintain anymore.
>
> The 2D view is [...]

I really like to use 2D view for arranging object, so I was missing it. I was not aware of "1" and "2" hotkeys option in 3D view, which replaces 2D view.
The option of "camera view gizmo" should be more visible. The on-screen buttons like existing 3 on the left would be great option.

sschwartzman, 2019-07-03:
I know this is an old-ish thread, but I just recently encountered this 'feature' and feel that I have a good use case to offer toward having it added back.

I use(d) the 2D view specifically to see if there are any gaps the layers that are printing after slicing, to look for defects in the design or scaling. Being able to shift up and down over a few layers, I'm often able to identify those defects, or at least confirm what I see in the 3D views. If you have a better way to perform such an exercise, please let me know.

I also use it for when I'm trying to "choose" a nice face on either side of a print, and want to see what the top- and bottom-fill patterns render to. I can do it in 3D mode fine, but it is easier to look at in straight 2D. That's more nice-to-have than my former use for it though.