prusa3d/PrusaSlicer #1202: Disregard Machine Limit settings and use default firmware settings

brotherchris, 2018-09-06:
### Version
1.41.0

### Operating system type + version
Win 10 64

### Behavior
The machine limit settings in the printer settings tab override the default firmware settings on the printer with Gcode commands at the start of the print. 

Is there away to ignore the machine limit settings in slic3r and use the default firmware options instead?

bubnikv, 2018-09-07:
Why would you do that? The machine limits are there to keep the firmware
and the slic3r's time estimator on the same page.

brotherchris, 2018-09-07:
Because I have been using slic3r for years as a non-propriotary slicer and use it for about 30 different printers and they are not all Prusa. Now I would have to update all the settings on every machine and the new ones coming in before I get my first print out.

Going forward is it Prusas intent to only allow Prusa printers the use of slic3r?

Thanks for the response.

bubnikv, 2018-09-07:
If you are not going to fill in the machine limits page, the print time
estimates will be off. There is nothing prusa specific here. If you don't
care about the print time estimates, switch slic3r to the reprap firmware
flavor.

brotherchris, 2018-09-07:
That works for me. Thanks again.

https://github.com/

<https://github.com/prusa3d/Slic3r/issues/1202#issuecomment-419321552>, or mute
the thread