prusa3d/PrusaSlicer #1203: Build Wx-0.9932 fail with VS2013, Is there anyone can provide <slic3rPE>/local-lib to me?

scott0122, 2018-09-06:
I'm build Slic3r PE following steps on https://github.com/prusa3d/Slic3r/blob/master/doc/How%20to%20build%20-%20Windows.md
But failed on building Wx-0.9932

Is there anyone can provide <slic3r>/local-lib to me?

vojtechkral, 2018-09-07:
What is the error?

bubnikv, 2018-09-07:
> But failed on building Wx-0.9932

With the binary packages provided, you should not need to compile wxwidgets
at all. Everything is precompiled already.

scott0122, 2018-09-07:
I see. I found that all lib required are precompiled in wperl64d-5.24.0-2018-03-02.7z's site/lib.
So it is not necessary to run 'perl Build.PL' & 'perl Build.PL --gui' ?

bubnikv, 2018-09-07:
I am not quite sure now, it has been a long time since I set up my
development environment. They should not hurt.

Please make sure you have your environment variables set by running
the C:\wperl64d\cmdline_init.bat

scott0122, 2018-09-07:
Finally, I successfully built the slic3r PE.
If you build Slic3r PE using wperl64d-5.24.0-2018-03-02.7z and slic3r-destdir-64.7z, you should not execute "perl Build.pl" and "perl Build.pl --gui". It will retrieve all necessary libraries from cpan.org again and cause wx build errors.

And the following files should be saved in "utf-8 with BOM" format.  Otherwise it will cause the cl.exe compilation source with errors. My system locale is Taiwan.
<Slic3r-PE>\xs\src\clipper.cpp
<Slic3r-PE>\xs\src\libslic3r\ExtrusionSimulator.cpp
<Slic3r-PE>\xs\src\libslic3r\GCode.cpp
<Slic3r-PE>\xs\src\slic3r\GUI\GUI.cpp
<Slic3r-PE>\xs\src\eigen\Eigen\src\Core\arch\CUDA\Half.h
<Slic3r-PE>\xs\src\libslic3r\PerimeterGenerator.cpp
<Slic3r-PE>\xs\src\libslic3r\PrintConfig.cpp
<Slic3r-PE>\xs\src\libslic3r\PrintObject.cpp
<Slic3r-PE>\xs\src\libslic3r\GCode\SpiralVase.cpp
<Slic3r-PE>\xs\src\libslic3r\SupportMaterial.cpp
<Slic3r-PE>\xs\src\slic3r\GUI\Tab.cpp
C:\local\slic3r-destdir-64\usr\local\include\boost-1_63\boost\spirit\home\support\char_encoding\iso8859_1.hpp

I think this document(https://github.com/prusa3d/Slic3r/blob/master/doc/How%20to%20build%20-%20Windows.md) should be updated.

vojtechkral, 2018-09-10:
@scott0122 Thanks, I'll update the Windows build doc.

> And the following files should be saved in "utf-8 with BOM" format. Or it will cause the cl.exe compilation errors. My system locale is Taiwan.

UTF-8 with BOM is problematic, if I recall correctly, the problem was that it makes MSVC compiler interpret the string literals and transform them according to the system locale. We don't want that, and so the current solution has been to save source files in UTF-8 _without_ BOM and have MSVC copy the strings verbatim (byte by byte), which we can then interpret in the l10n layer as UTF-8.

What errors have you been getting with respect to encoding?

scott0122, 2018-09-11:
1>C:\local\Slic3r-PE\xs\src\slic3r\GUI\GUI.cpp : warning C4819: The file contains a character that cannot be represented in the current code page (950). Save the file in Unicode format to prevent data loss
1>C:\local\Slic3r-PE\xs\src\slic3r\GUI\GUI.cpp(340): error C2001: newline in constant
1>C:\local\Slic3r-PE\xs\src\slic3r\GUI\GUI.cpp(342): error C2144: syntax error : 'void' should be preceded by ')'
1>C:\local\Slic3r-PE\xs\src\slic3r\GUI\GUI.cpp(342): error C2448: 'dots' : function-style initializer appears to be a function definition
1>C:\local\Slic3r-PE\xs\src\slic3r\GUI\GUI.cpp(350): error C2065: 'dots' : undeclared identifier
1>C:\local\Slic3r-PE\xs\src\slic3r\GUI\GUI.cpp(351): error C2065: 'dots' : undeclared identifier
1>C:\local\Slic3r-PE\xs\src\slic3r\GUI\GUI.cpp(355): error C2065: 'dots' : undeclared identifier
1>C:\local\Slic3r-PE\xs\src\slic3r\GUI\GUI.cpp(892): error C2065: 'dots' : undeclared identifier
1>C:\local\Slic3r-PE\xs\src\slic3r\GUI\GUI.cpp(892): error C2664: 'wxButton::wxButton(const wxButton &)' : cannot convert argument 3 from 'const wxPoint' to 'const wxString &'
1>          Reason: cannot convert from 'const wxPoint' to 'const wxString'
1>          No user-defined-conversion operator available that can perform this conversion, or the operator cannot be called
5

scott0122, 2018-09-11:
@vojtechkral After saving thoese files in utf-8 with bom format by ultraedit, All compilation errors are gone.

bubnikv, 2018-09-12:
Cool. Closing then.

vojtechkral, 2018-09-12:
(I'd like to leave this open to remind myself to update the doc and perhaps fix the GUI.cpp encoding problem.)

scott0122, 2018-09-12:
@vojtechkral Not only GUI.cpp. Some of them are compilation errors, some of them are just warning.

bubnikv, 2018-12-12:
We got rid of Perl, so there will be no more issues with compiling Wx-0.9932
Closing.

closed 2018-12-12 by bubnikv