prusa3d/PrusaSlicer #1207: MMU - Change Temperature/Material Purge Order Issue

hanswwilson, 2018-09-08:
### Version
1.41.0+win64

### Operating system type + version
Windows 10 Professional

### Behavior
When printing an object using multiple dissimilar materials that use different print temperatures the order in which the temperature change, material change, and purge event becomes important to prevent extruder jams. Currently the behavior of multi-material sliced objects using different print temperatures proceeds with the following behavior:

Material change begins - Print temp is changed - Material is changed - Purge occurs - Printing Resumes

By the time the purging occurs the extruder will be at the temperature required to print the new material. This order is fine if you are going from a lower temperature material to a higher temp one as the lower temp material is purged at the higher printing temp. However, if you change from a high temp material to a low temp one, since the purge occurs after the temperature has changed, it will try to purge a material from the extruder while using a temperature far below its normal printing temperature. As an example printing PLA at 205c and ABS at 255c it will potentially be trying to purge ABS from the nozzle at 205c while swapping from the ABS to the PLA.

This behavior will occur for any object sliced for multi material using materials with dissimilar print temperatures.

### Expected Results
During the gcode generation of the material change procedure it should check whether or not it is changing from a lower print temp to a higher one, or a higher temp to a lower one. It should then change the ordering of the purge and temperature change so that the purge always occurs while the extruder is at the hotter print temperature.

Higher temp to low temp order:
Change material - Purge - Change print temp

Lower temp to high temp order:
Change print temp - Change material - Purge

### Actual Results
When printing and changing materials it always follows the same order of 
Change print temp - Change material - Purge

[MMUTempChangeExample.zip](https://github.com/prusa3d/Slic3r/files/2363008/MMUTempChangeExample.zip)

jfestrada, 2020-10-22:
Hi, 

This issue still present in the 2.2.0 release and is very disapointing. In my case I have had issues even with different types of PLA whose works at different temperatures from 205ºC to 215ºC. To my mind is easy to fix even with a python script. 

Br
Juanfra

Unismo, 2020-10-22:
Hello,

The same goes for the latest firmware version, many problems with different PLAs like 870, PLA + and normal PLA in the same print, many errors in the loads when the temperature is different between filaments.

amarfil, 2020-10-22:
Hi,

I have this issue too in version 2.2.0. Mixing different filament brands causes loading filament problems on color changes because generally they need different temperatures. Please consider fix it soon.

Thanks

amatulic, 2023-02-13:
I believe I have fixed this. I haven't raised a PR yet (and frankly, with the backlog of pull requests I see, I'm not confident that this fix would make it into a release).

The fix required some minor changes to `/src/libslic3r/GCode/WipeTower.cpp` in the functions `toolchange_Unload()` and `toolchange_Wipe()`.

It would work like this:
- If changing to a hotter filament, the temperature change is started when the cooler filament unloads and the new hotter filament loads. It waits (if needed) for the temperature to be reached before starting the wipe on the wipe tower. After the wipe is done, the print resumes.
- If changing to a cooler filament, there is no temperature change during unloading and loading, the temperature is kept hot. The lower temperature target is set when wiping starts, so that the nozzle purges the hotter filament at a higher temperature and cools during the wipe. When the wipe is done, it waits (if needed) for the lower temperature to stabilize before the print resumes.

rasssta, 2023-08-07:
Great work @amatulic!

Anyone knows of a way to get Prusa to actually look at this issue, and its linked PR? Seeing as the issue is from 2018 it feels like something has to be done to get their attention to it.

Jackjan4, 2023-08-15:
Today I printed PLA + TPU on the MMU3 and can now understand the problem. This really needs to be fixed for the MMU3 to be a real multi **material** solution.

Mike8-I, 2023-08-30:
@Prusa-Support: this is a really needed feature. It makes the  MCU (multicolour unit) that it is at the moment really a multi material unit (MMU) as the name says and what I bought it originally for, years ago. It never really worked. I will give it another try with the 3.0 firmware. Then I need the multi material ability.

amatulic, 2023-08-30:
@josefprusa I have been using the fix in PR #9737 for several months to print PLA with PETG support using the soluble support preset. It's been working well. Now that the MMU3 is out, it really needs to be a **multi-material** unit, not just a multi-color unit, and PR #9737 accomplishes that, simply by using the correct temperature on the wipe tower.

It's a really simple fix for a long-term problem. Without the fix in PR #9737, you get a jammed nozzle when switching to a cooler-temperature filament.

michaelbryan, 2024-06-24:
@josefprusa Just echoing the above. Lots of MMU3s are rolling out. Lots of us want real multi material solutions.

JGeiger2024, 2024-10-07:
Anyone know whether there is a fix in progress by Prusa?

Cinderhaze, 2025-02-18:
With all the multicolor changes In alpha 2.9.1, I was sad to this this not included in the set of fixes!

closed 2026-09-29 by prusaslicer-bot[bot]

Cinderhaze, 2026-09-29:
Do we know if this is still present in 3.0?

amatulic, 2026-09-29:
@Cinderhaze -- I plan to test it, and open another PR if needed. This PR has undergone so many PrusaSlicer version changes that it couldn't even be merged anymore, even though there were no merge conflicts. I'll start a fresh one, but I was planning to wait until after the "alpha" version.

amatulic, 2026-10-23:
This should be re-opened. @hanswwilson would you do so?

hanswwilson, 2026-10-23:
I don't have permission to re-open the issue. The button is greyed out.

[image]

amatulic, 2026-10-23:
That's a shame. You created this over 8 years ago, I fixed it over 3 years ago, it was a fairly trivial change, and it never got merged. Oh well. I'll test 3.0.0 and see if the same fix still needs to be applied.