prusa3d/PrusaSlicer #1208: Weird artifacts as of this release 1.41.0 x64

akash1769, 2018-09-08:
### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions_1.41.0 x64

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_ Win 10 Pro latest

### Behavior
* _Describe the problem_ artifacts showing up that look similar to support material showing up randomly
* _Steps needed to reproduce the problem_ Download Wavey heart (1st version) from thingiverse choose .2 linear advance. No top layers. No fill. Scale 50%.  Slice. Change layer thickness .01 up or down. Seems to be least  weird 
[image]
[image]

at .15
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
[config.zip](https://github.com/prusa3d/Slic3r/files/2363031/config.zip)
[waveylove1.zip](https://github.com/prusa3d/Slic3r/files/2363034/waveylove1.zip)

FidelCapo, 2018-09-10:
@akash1769 These artifacts you describe are not supports but solid infill. It appears there because you have the feature "Ensure vertical shell thickness" that tries to keep wall thickness by adding infill.

Was this explanation helpful to you? Can I close this issue?

akash1769, 2018-09-11:
You the man! Thanks

Yes you may close da issue