prusa3d/PrusaSlicer #1216: Machine limits reset when 'Supports Silent Mode' is toggled

nessatse, 2018-09-09:
### Version
Slic3r Prusa Edition
Version: 1.41.0+win64
Build: Slic3rPE-1.41.0+win64-full-201809010758
Operating System: MSWin32
System Architecture: MSWin32-x64-multi-thread
Windows Version: Microsoft Windows [Version 10.0.17134.228] 

### Behavior

Whenever you toggle 'Supports silent mode' under Printer Settings->General, all the settings under 'Machine Limits' are reset to their built in values.

bubnikv, 2018-09-27:
@YuSanka would you please look into this issue? Thanks.

Enverex, 2019-04-18:
Just ran into this issue myself and luckily noticed before printing anything.

So here are my original values:
[image]

Disabling silent mode then results in this (which wasn't either of the value sets from earlier):
[image]

Then re-enabling the option **doesn't** restore the original values, it adds the silent set back, but leaves the default changed values in the normal boxes:
[image]

YuSanka, 2019-04-18:
Thanks for your report. We know about this bug  and I’m working on it.

bubnikv, 2019-05-16:
This should be fixed in the PrusaSlicer 2.0.0-rc. Closing.
Feel free to reopen if you still believe there is an issue,.

closed 2019-05-16 by bubnikv