prusa3d/PrusaSlicer #1217: [FEATURE REQUEST] Display count of tool changes

pleppik, 2018-09-09:
### Version
1.41.0+

### Operating system type + version
All

### Behavior
The single most important factor in optimizing multimaterial prints is minimizing the number of tool changes.

Currently there does not appear to be any way to see how many tool changes there are in the sliced model.

It would be very useful to display the number of tool changes after slicing the model to make it easy to see how changing print settings and/or model arrangement will affect the number of tool changes.

bubnikv, 2018-09-11:
@lukasmatena would you please add this statistics? Thanks.

lukasmatena, 2018-09-19:
@pleppik Thank you for the suggestion. The number of tool changes during a MM print is now shown in 'Sliced Info' box (implemented into master branch by https://github.com/prusa3d/Slic3r/commit/712fef0669a491c4335286d61d34ebb0b24345e6).
Closing the issue.

closed 2018-09-19 by lukasmatena