prusa3d/PrusaSlicer #1218: Implement DPI scaling support

rackley096794, 2018-09-10:
### Version
1.41.0

### Operating system type + version
Windows 10 1809  (same issue is present on earlier versions)

### Behavior
* When using high resolution monitors, the user interface is fuzzy
* To repro, set monitor scaling to something larger than 100% (e.g. 125%, 150%, etc).
* Expected results are that the UI is correctly and cleanly rendered
* Actual result is the UI is fuzzy

DPI scaled at 125%:
[image]

DPI scaled at 100%:
[image]

See this link for more information:
[High DPI Desktop Application Development on Windows](https://docs.microsoft.com/en-us/windows/desktop/hidpi/high-dpi-desktop-application-development-on-windows)

daktylus, 2018-09-10:
+1

and adding the request to use a higher resolution picture of the Printbed (It was better in an older version but it was lately swapped for a low-res 512x512 pixel image. Name: "mk2_top.png")

Artem-B, 2018-09-24:
Until this is fixed, the work-around is to right-click on slic3r.exe, go to `Properties`, select `Compatibility` tab and click on `Change high DPI settings`. At the bottom check `Override high DPI scaling behavior` and select `Application` from the drop-down menu.

After that the app will be drawn without scaling. The down side is that manus and text will also be unscaled and may be too small if the scaling factor was high enough.

bubnikv, 2019-04-13:
Implemented in https://github.com/prusa3d/Slic3r/releases/tag/version_1.42.0-beta1

closed 2019-04-13 by bubnikv