prusa3d/PrusaSlicer #1219: Upload to Printer Host - naming dialogue issue

DrJuJu, 2018-09-10:
### Version
1.41.0 win64

### Operating system type + version
Windows 10 version 1803

### Behavior
The dialogue for naming the file that is sent to the printer, by default, only shows the last nine characters plus ".gcode", even though the dialogue box has enough space to show more of the filename. see attached. 

I applaud the addition of layer height, material etc. but it would be nice to see the full file name.

[image]

DBa2016, 2018-09-11:
Workaround: press "Home". The whole file name is there, it is just not immediately visible.

vojtechkral, 2018-09-11:
I've got a fix for this, it's just not yet released...

commit 4fa87490 (2018-09-13): Print host improvements: presistent path & start print, bugfixes
Fixes #1219
Fixes #1004
Fixes #1106
Fixes #1173
commit 81da613c (2019-02-18): Printhost: Persist upload path & start print checkbox (re-add lost code)
Fixes #1219
Fixes #1004
Fixes #1106
Fixes #1678
commit 08f1459a (2019-02-19): Printhost: Persist upload path & start print checkbox (re-add lost code)
Fixes #1219
Fixes #1004
Fixes #1106
Fixes #1678
closed 2019-02-19 by vojtechkral