prusa3d/PrusaSlicer #1220: Layer editing not working

gime2012, 2018-09-10:
### Version
Slic3rPE-1.41.0+win64-full-201808301410

### Operating system type + version
Microsoft Windows (MSWin32)  [Version 10.0.17134.228] 

### Behavior
I am unable to select layer editing since it is greyed out. I have enabled variable layer height in Printer Settings tab and have set the min layer height at 0.07 mm and max layer height at 0.25 mm. I select the object on the plater and am able to use all the buttons like scale or split except the Layer editing button.
I have used variable layer height feature sucessfully in the past.

* _Steps needed to reproduce the problem_

* _Expected Results_
To be abele to use variable layer height feature like I was able in the past.
* _Actual Results_
[image]

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

FidelCapo, 2018-09-11:
@gime2012 Unfortunately, I'm not able to reproduce this issue. Can you please send me config.ini ? Thank you

bubnikv, 2018-09-11:
The layer editing is only enabled for OpenGL 2.0 and newer. What OpenGL
version does Slic3r report you when you execute menu -> System Info?

bubnikv, 2018-09-11:
> Sure @FidelCapo <https://github.com/FidelCapo> here is the config
> PETG.zip <https://github.com/prusa3d/Slic3r/files/2370578/PETG.zip>
>
> @bubnikv <https://github.com/bubnikv> my OpenGL is1.1, I'll update it and
> report if that had helped

>  my OpenGL is1.1, I'll update it and report if that had helped

Then no wonder that the layer editing is not working. It requires OpenGL
2.0 or newer.

gime2012, 2018-09-11:
I have updated the OpenGL and now it works. Thank you!

closed 2018-09-11 by gime2012