prusa3d/PrusaSlicer #1221: Custom extension in output_filename_format not working properly in UI

zmast, 2018-09-11:
### Version
1.41.0+

### Operating system type + version
macOS 10.13.6

### Behavior
* _Describe the problem_
When the parameter output_filename_format contains a custom file extension, for instance `.hvs`, the "save as" dialog will still force the extension to be `.gcode`, resulting in a file named `filename.hvs.gcode`.

* _Steps needed to reproduce the problem_
  * In _Printer Settings > Output options > Output filename format_ enter `[input_filename_base].hvs`
  * Slice a model and click _Export G-code..._

* _Expected Results_
The dialog should use a filename like `my_model.hvs`.

* _Actual Results_
The dialog uses a filename like `my_model.hvs.gcode`.
Note: even if you delete the `.gcode`, it gets added again when saving the file.

_Is this a new feature request?_
No

bubnikv, 2018-09-11:
What is the .hvs extension?
I suppose the input field expect you to give the file name without the
extension. Why do you want to change the extension?

zmast, 2018-09-11:
My 3D printer, a Cubicon Style, requires files with that extension (plus some custom G-CODE), it won't see .gcode files. The .hvs extension is probably named after the manufacturer (HiVision Systems).
So, having Slic3r save the output file with the correct extension avoids having to rename the file manually.

The default value for that field is `[input_filename_base].gcode`, so one would expect that changing the file extension is possible.
I suppose the easy fix would be to add the hvs extension to `FILE_WILDCARDS["gcode"]`, although this doesn't address the generic case.

bubnikv, 2018-12-12:
The upcoming Slic3r 1.42 alpha will maintain your custom extension.
Closing.

closed 2018-12-12 by bubnikv

marcoburato, 2023-08-19:
@bubnikv This issue is now happening again on 2.6.0. Can you reopen the ticket?

marcoburato, 2023-10-07:
Just confirming this was indeed fixed in 2.6.1. Many thanks.