prusa3d/PrusaSlicer #1222: The configuration aren't saved

blithe74, 2018-09-11:
### Version
1.41.0+win64

### Operating system type + version
Windows 7 Professional, SP1, x64, Russian

### Behavior
During start of the program (slic3r.exe) appears  the window "Slic3r error":
[image]

The folders:
C:\Users\Internet\AppData\Roaming\Slic3rPE\filament\
C:\Users\Internet\AppData\Roaming\Slic3rPE\
C:\Users\Internet\AppData\Roaming\Slic3rPE\printer\
already exist, but empty. User role - simply user (non Administrator) have the rights for all folders: create, read, write, delete.

After changing the profile in windows "Print settings" "Filament settings", "Printer settings" and to push to icon "Diskette", the  programm did not save settings (folders is empty).

bubnikv, 2018-09-12:
Did you redirect the "C:\users\internet" folder somehow using Windows redirect point?

blithe74, 2018-09-12:
Yes link to folder. But from Menu of program -> the folders is displayed correctly, and original Slic3r work properly.
May be need use another Windows application function or check the Path to really folder or link. You can insert debug message in program from many Windows application function, and I send result to You.

С уважением, Александр Шуковилов

bubnikv, 2018-09-13:
> Yes link to folder.

How exactly did you create the link? What kind of link? I know around three
options: The Explorer shortcut, the hard link and the soft link.

> But from Menu of program -> the folders is displayed correctly,

Sure, that is basically the Windows Explorer in action, which interprets
the redirections correctly.

> and original Slic3r work properly.

The original Slic3r still uses Perl in this code.

There is a long standing issue with the boost library that we are using now
in regard to interpreting windows reparse points. Just search for the
"reparse" word in our issue tracker, there are around two issues like yours
entered already. We may need to invent some workaround.

bubnikv, 2018-09-13:
These issues may be related:
https://github.com/prusa3d/Slic3r/issues/1123
https://github.com/prusa3d/Slic3r/issues/732

bubnikv, 2018-09-13:
Then it is a duplicate of #732, closing this one.

I am still waiting for the ticket https://svn.boost.org/trac10/ticket/10900 to be implemented and closed.

bubnikv, 2018-09-13:
I really don't need to see your errors, I bet I can reproduce it with mklink. It's just that we don't currently have the resources to dig into this issue and it is pretty marginal.

closed 2018-09-13 by bubnikv