prusa3d/PrusaSlicer #1225: Slicing bug : infill go through perimeter on a simple object !

sebas76, 2018-09-12:
### Version
Slic3rPE-1.41.0+win64

### Operating system 
Windows 10 Pro 64 bits - Version 1803

### Behavior
Slicing a simple object (a disk with 3 holes) and get bad result : the infill go through and beyond the perimeters as you can see here :
[picture](https://user-images.githubusercontent.com/21216559/45428717-1a7a9b00-b6a2-11e8-8d72-e388b929aca2.jpg)
I have checked on slic3r and s3d without any problem.
I have tested many different presets but the result is always the same.
I get this issue on Slic3r PE 1.41 and 1.40.1 too.

#### STL/Config (.ZIP) where problem occurs
[issue.zip](https://github.com/prusa3d/Slic3r/files/2375372/issue.zip)
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

BerndJM, 2018-09-13:
Have a look at your "Infill/perimeters overlap" setting (Print Settings -> Advanced) 
There's a missing **%** sign so your overlap is set to 25**mm** ...

bubnikv, 2018-09-13:
Thanks @BerndJM for your investigation. 
@YuSanka would you please add a validation to this parameter, so one will not enter an excessive number by mistake? Thanks.

sebas76, 2018-09-13:
Thanks @BerndJM.
I haven't seen my mistake before... All my apologizes...
This was the wrong parameter, it's missing % in overlap.

bubnikv, 2018-09-27:
> I haven't seen my mistake before... All my apologizes...

You rightly noticed us to improve the software. We shall test for excessive values everywhere in Slic3r, where the dual value / percent entry is expected.

commit 17164ee3 (2018-12-17): Fixed #1225 (Added "Parameter validation" for "mm or %" values)

+ fixed correct updating of the "Contact Z distance" parameter
closed 2018-12-17 by YuSanka