prusa3d/PrusaSlicer #1226: Platter is spelled wrong

cyrusjk, 2018-09-12:
### Version
1.41.0

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Ubuntu 18.04
### Behavior

In english, the word "Platter" has two "t"s, not one.

_Is this a new feature request?_
Nope.
#### STL/Config (.ZIP) where problem occurs

Zingabopp, 2018-09-13:
It's not a typo, plater is the word they use in the original Slic3r. It's not exactly a word from the dictionary, but it seems commonly used for 3D printing (The thing you use to set up the build plate = Plater).

vojtechkral, 2018-09-17:
See also: https://github.com/Rhoban/Plater

cyrusjk, 2018-09-24:
Well then I guess I'll just shut up then. ;)

closed 2018-09-24 by cyrusjk

bubnikv, 2018-09-25:
I was confused as well, I did not find a reasonable explanation of the term in any dictionary.
https://en.wiktionary.org/wiki/plater
https://www.dictionary.com/browse/plater?s=t
https://www.merriam-webster.com/dictionary/plater

zanderk, 2019-04-06:
The "somebody in the past misspelled it so that's the way it is" excuse seems pretty lame.
Software types write code not dictionary's. 

    platter: A large shallow dish or plate, used especially for serving food.
seems a lot better definition of what the part sits on than

    plater: someone who plates metal.

vojtechkral, 2019-04-08:
It says [on wiktionary](https://en.wiktionary.org/wiki/plater):

> plater: (biology) A device for depositing cells on a plate.

Seems related. Granted, slicing isn't biology, but the purpose seems to be pretty much the same to me.

That said, I don't really care what colour the bike shed is painted so to speak :)

Kaelum, 2024-09-13:
These definitions are from the Merriam Webster Dictionary:

`Plater` is a noun:
1 : one that [plates](https://www.merriam-webster.com/dictionary/plates)
2 a : a horse that runs chiefly in [plate](https://www.merriam-webster.com/dictionary/plate) races
  b : a racehorse that competes in the lowest grade of races

`Platter` is a noun:
1 a : a large [plate](https://www.merriam-webster.com/dictionary/plate) used especially for serving meat
  b : [plate sense 3c(2)](https://www.merriam-webster.com/dictionary/plate)
2 : a phonograph record

Thus `Platter` is the correct word.

cyrusjk, 2024-09-13:
This was answered years ago, and the initial explanation was sufficient.

Kaelum, 2024-09-13:
> This was answered years ago, and the initial explanation was sufficient.

Well, the spelling is still incorrect in the UI.  ;)