prusa3d/PrusaSlicer #1228: Lift nozzle when moving over a support structure

txgruppi, 2018-09-13:
### Version
1.41.0+

### Operating system type + version
OS X 10.13.6

### Behavior
* Description: The printer should lift the nozzle when traveling from support to support (video below)
* Steps: Create a structure that uses support and let Slic3r make the nozzle travel from one part of the support to another part of the same support
* Expected: The nozzle should lift to not touch the support material
* Actual: See attached video

Is this a new feature request? No

#### STL/Config (.ZIP) where problem occurs
In the video I'm trying to print the biggest of the 3 parts in the STL file, and with a different orientation.
https://www.thingiverse.com/download:419864

**WARNING: LOUD MACHINE SOUNDS**

You can hear the moment the nozzle moves over the existing support.

https://youtu.be/YfzORXdYeVM

bubnikv, 2018-09-27:
Why don't you enable Z-hop?

txgruppi, 2018-10-03:
Where is this option? I could not find it.

bubnikv, 2018-10-03:
printer settings -> Extruder 1 -> Retraction -> Lift Z

txgruppi, 2018-10-03:
I have it set to `0.2`.

bubnikv, 2018-10-05:
> I have it set to 0.2.

Maybe you need to increase it? For Prusa i3 MK3 we use 0.6 mm

txgruppi, 2018-10-05:
It is weird because the problem doesn't happen for the print, only for the supports.

bubnikv, 2018-10-05:
Likely because the supports are fragile and sparsely connected.

bubnikv, 2018-12-12:
@txgruppi Any update? Did increase of the z-hop help?

txgruppi, 2018-12-12:
Sorry, I didn't work with any print yet that is similar to the one that had this problem.

closed 2018-12-12 by txgruppi