prusa3d/PrusaSlicer #1230: Inexact rotations when using the new rotation gizmo

Quintox303, 2018-09-14:
### Version
1.41.0+
### Operating system type + version
Mac OS High Sierra 10.13.6

### Behavior
* When the new rotation gizmo ported from Prusa Control is used to rotate a part, and the 45 degree snapping increments are used, the results are not an exact 45 degree rotation. This can be seen if you then right click on the part and go to the rotate > z axis menu. Attached is a screenshot showing the long number of decimal places and imprecision.

* _Steps needed to reproduce the problem_
1. Import a STL file
2. Click on the rotation gizmo
3. Rotate the part by 45 degrees using the snapping increments
4. Right click on the part and navigate to the rotate > z axis menu

* _Expected Results_
Exact snapping in 45 degree increments, so the attached screenshot would only read "135"

* _Actual Results_
  
[image]

lukasmatena, 2018-09-15:
Floating point calculations are inevitably not exact and small differences from mathematically correct solution frequently appear in the low-order digits. Unless the error sums up quickly or produces numerical instabilities (and I cannot imagine that happening it this case), there is nothing to fix as far as I'm concerned. What we could do is to show the actual angle with smaller precision (for example to two decimal places).
Can you confirm that the problem does not manifest in any other way?

Quintox303, 2018-09-15:
Thank you for the explanation Lukas! That makes a lot of sense. I think that showing the angle to two decimal places would be a good idea. I didn't find any other issues with the rotation :)

Looxoor, 2018-09-19:
Another suggestion would be to trunc the float if the user has triggered the set angles.
That way you get the correct angles either way, even if it's just in the background.
I know it would bother me knowing it's not a clean number.

Just my two cents.

bubnikv, 2018-12-12:
The upcoming slic3r 1.42 alpha does not show this kind of issue. Closing.

closed 2018-12-12 by bubnikv