prusa3d/PrusaSlicer #1236: M73 is SAVE MODAL STATE

arhi, 2018-09-17:
### Version
1.41.0

### Operating system type + version
Irrelevant

### Behavior

Slic3r adds M73 codes into output gcode file using M73 code in a way it's not intended by cnc machines and documented by NIST. M73 code is used to SAVE and autorestore MODAL STATE when scripting or OUTER CHUCK CLAMP when not. Why chose M codes that are already well defined and used by the community instead of using free M codes. You can use M573 for e.g. 

#### STL/Config (.ZIP) where problem occurs
irrelevant

bubnikv, 2018-09-18:
https://reprap.org/wiki/G-code#M73:_Set_build_percentage

bubnikv, 2018-09-27:
Thanks for comment, but the M73 has been introduced by MakerBot for this purpose, we have already adapted it, and it is late for us to change the decision.

Closing your issue.

closed 2018-09-27 by bubnikv