prusa3d/PrusaSlicer #1237: Load Part Parameters have no affect

Avatarsia, 2018-09-17:
### Version
1.41.0+win64
I create a Tree Support with Meshmixer and Load it via the double click menu to my objekt.
After i set the parameter and infill speed to 1 mm/s because they always break during the print.
I also gave them just 2% infill and 1 paramter.

End of the story, the supports were printed with the same details like the rest of the Objekt.

[issue.zip](https://github.com/prusa3d/Slic3r/files/2389972/issue.zip)

FidelCapo, 2018-09-18:
Unfortunately, I am unable to reproduce this issue. I attach a 3mf file where I have set the same parameters that you describe and it seems to me that everything works.
[goku_kid.3mf.zip](https://github.com/prusa3d/Slic3r/files/2392693/goku_kid.3mf.zip)

Avatarsia, 2018-09-18:
so far i can see you didnt tried the reduced speed, right ?

FidelCapo, 2018-09-18:
I tried. This parameter is added to part of the "goku_kid support" model. Speed - small perimeters. In the gcode export speed preview, you can see the speed limit on this part.

Avatarsia, 2018-09-18:
thats why i asked, everythink is nice and yellow so about 60 mm/s

FidelCapo, 2018-09-18:
[image]
If the 3mf file I've attached I'm exported, it's clearly visible on the preview that the support is printing at a speed of 1mm / s.

Avatarsia, 2018-09-18:
well i forgott to export in gcode, my fault , yeah u r right, thanks

FidelCapo, 2018-09-18:
You are welcome. So, can I close this issue?

Avatarsia, 2018-09-18:
yes please do so