prusa3d/PrusaSlicer #1239: Fusion 360 to Slic3r

davidewen, 2018-09-18:
### Version
_Slic3r Prusa Edition_

_Use `1.41.0+` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
macOS High Sierra v10.13.6

### Behavior
* Clicking the "make" button in Fusion 360, then setup output to 3D print utility and Slic3r, hit ok and nothing shows up in Slic3r. 

Works great with Simplify3D. 

I would use S3D, but the settings for the Prusa i3 MK3 aren't as good as S3r

bubnikv, 2018-09-18:
Sorry, but I don't understand your workflow, as I am not myself a user of Fusion 360. Would you be more specific as what kind of functionality you are missing from Slic3r? Some command line parameters?

Maybe a --gui parameter is what you are looking for. Without the --gui parameter Slic3r will slice on background.

dikodahan, 2018-09-18:
When a design is complete in Fusion 360 and you are ready to export it as an STL file, Fusion supports an option to export it and immediately open your favorite slicing app with this STL file loaded already. This works perfectly with Meshmixer, Simplify3D, Cura and others. However, when you choose Slic3r, it does open Slic3r, only not STL file is actually loaded in it.

This is a huge time saver with other slicing apps!
Hope this clarifies the issue.

Phydam, 2018-09-18:
What version of slicer and fusion 360 are you using. I just tried it using slicer 1.41.0+ and Fusion360 2.0.4391 and it worked.  Which slicer executable did you select?

Zingabopp, 2018-09-18:
Also worked for me on Slic3r PE 1.41.0 (Windows 10). Just selected Slic3r.exe (not the console or noconsole ones), maybe this is a MacOS specific issue.

dikodahan, 2018-09-18:
I am using Slic3r 1.41.0+ final version and the latest Fusion 360 2.0.4567
I just tried it again to validate the again, Slic3r opens, however, no STL file loaded to it.

And just in case it might be of relevance, This is running on MacOS High Sierra 10.13.6.

davidewen, 2018-09-18:
> What version of slicer and fusion 360 are you using. I just tried it using slicer 1.41.0+ and Fusion360 2.0.4391 and it worked. Which slicer executable did you select?

I'm using:
Slicer: Slic3r Prusa Edition v1.41.0+
Fusion 360: v2.0.4567
Computer: MacOS High Sierra v10.13.6 (17G65)

I have the same issue that @dikodahan is having - - Slic3r opens, but no STL is loaded.

Phydam, 2018-09-18:
Sorry, I'm on Windows 10

Looxoor, 2018-09-19:
It's not a Fusion problem, the same happens when you try to open an .stl with Slic3r PE from finder as well.

bubnikv, 2019-01-10:
On OSX, the file to open is not delivered through a command line parameter, but through an event. This has finally been implemented with c258375fe298083e2d10f270db5039d6ca7e3e04 

On Windows, there was a bug in the command line processing. Fixed with 5b1c1d592261e96f626b1172601a5bd0e4348575

Closing.

closed 2019-01-10 by bubnikv

davidewen, 2019-01-11:
Bubnikv,

Thanks for the response.

However, it’s unclear what your response means.
Am I supposed to do something with your reference to c258375?
And if so, what? There is no explanation what to do with it via the link.

Cheers,
David

c-stewart, 2019-01-11:
That is where they fixed the issue in code. You need to download the new
version of slic3r when it's released.