prusa3d/PrusaSlicer #1240: Slic3r incorrectly slices this STL file

timur-tabi, 2018-09-19:
### Version
1.41.0+

### Operating system type + version
Mac OS 10.11.6

### Behavior
This particular STL file slices correctly in Cura, but in Slic3r, some of the bottom pieces are truncated at an angle.  I can't really describe it, so I've attached screen shots to explain it.

[slic3r_bug.zip](https://github.com/prusa3d/Slic3r/files/2395332/slic3r_bug.zip)

timur-tabi, 2018-09-19:
Forgot to attach the generated gcode:

[soap_tray_0.2mm_PLA_MK3.gcode.zip](https://github.com/prusa3d/Slic3r/files/2395340/soap_tray_0.2mm_PLA_MK3.gcode.zip)

bubnikv, 2018-09-19:
According to @FidelCapo , your model is quite broken, and the NetFabb service did not manage to fix it. We will look into it, if Cura could slice it, we should as well, but it will take time.

timur-tabi, 2018-09-19:
I recreated my model in Sketchup using a completely different technique, and then exported it to STL and loaded it in Slic3r.  This new version is even worse.  I don't understand.  I'm no professional, but the item I'm making is not at all unusual.

And as before, it renders perfectly fine in Cura.

[soap_tray_2.zip](https://github.com/prusa3d/Slic3r/files/2397142/soap_tray_2.zip)

jkavalik, 2018-09-19:
I tried to load the STL to Meshmixer and it shows many non-manifold parts - the red edges mean that there are more than two triangles sharing the edge so there may be troubles deciding which side is INSIDE and which is OUTSIDE.

[image]

[image]

timur-tabi, 2018-09-22:
I don't understand how there can be any edges with more than two triangles.  First, all surfaces are rectangles, so I don't know why you say triangles.  And second, the entire thing is composed of rectangular cuboids: basically, just rectangles that have been push/pulled.

Anyway, I managed to resolve my problem by tediously deleting all the superfluous lines on all the surfaces (e.g. the red squares on the mesh in your second picture).

Feel free to figure out why Slic3r is confused with my STL.  You can close this ticket if you like.

jkavalik, 2018-09-22:
STL does not know rectangles, a mesh is a set of triangles (and lot of 3D software works with triangles too, at least internaly). But each rectangle can be just split to 2 coplanar triangles so that is just a detail. The problem here seems to be that your objects possibly intersect? The second case looks like long rectangles are crossing so side of one is partially inside the other, you could see that in eg. Meshmixer by moving the camera inside some object. Now as STL does not have any topology information (what triangles are connected together) the loader has to decide that just from points and triangles stored in the STL. For object  with a "manifold" geometry that is not a problem because in the original geometry each edge belongs to exactly two triangles. But if the original geometry is not manifold, in the STL file you can see 3 or more triangles with an edge defined over two points with the same position, and without any topology information nothing says which two triangles should be connected and which are just a coincidence. The loader then has to decide somehow (or just refuse to load the file, but that would be frowned upon by users..) and different decisions lead to different errors in the object. (there are other reasons for a geometry to be non-manifold, edge with only one triangle, so "open" to nowhere etc. but that does not look to be your case)

BirkBinnard, 2018-09-22:
The original file has many non-manifold edges as shown in red in jkavalik's post above. When loaded into Plic3r 1.41.0 it reports 2227 errors and 3D display looks weird. To fix the part I ran it through 3D Builder and it produced the file I have attached. This one slices just fine.

Unfortunately 3D Builder doesn't tell what the errors were that it fixed, but I've found that it fixes most problems with STL files. Of course there are certain types of problems it can't fix - but that's a different matter.
[soap_tray_2.zip](https://github.com/prusa3d/Slic3r/files/2408003/soap_tray_2.zip)

bubnikv, 2018-10-01:
Sorry, but this object is really badly formed, and the netfabb was not able to make a sense of it.
Kudos to Cura, that it could handle it somehow.

closed 2018-11-20 by timur-tabi