prusa3d/PrusaSlicer #1241: Duet - start simulation after upload

mloidl, 2018-09-19:
### Version
1.41.0

### Operating system type + version
All

### Behavior
A Duet running reprapfirmware is able to start a simulation run instead of start a print.
When uploading a file to duet it should be possible to tick a box "Start simulation after upload" instead of the already existing checkbox "Start print after upload".

@vojtechkral 
I'd like to implement this feature because it was requested from one of the members in the duet forum.
I saw that you already did some refactorings in this area in the branch `vk-octoprint`

**Some questions:**
- Do you have a timeline when your  `vk-octoprint` branch will be merged into `master`?
- Would it be ok for you when i implement this feature based on your `vk-octoprint` branch and create a pull request directly to this branch?
- In `PrintHostSendDialog.hpp` the variable `bool can_start_print` is not used any more. The idea when i introduced this flat, was set only actions visible which are allowed for the Host. For `can_start_print` it's currently not needed because both implementations (Duet and Octoprint) are able to do this. For starting simulation i'd like to introduce a new flag `bool can_simulate_print` which is set to true only for Duet.
- Since both actions are not be possible at the same time i'd like to implement a locking mechanism. So only one or no action can be selected.
- To fit the requirement #1106 i would add a new config-key `CONFIG_KEY_SIMULATE` to save the last value analog to the start print checkbox

Would this ok for you?

_Is this a new feature request?_
yes

mloidl, 2018-09-20:
@vojtechkral 
I've created commit https://github.com/mloidl/Slic3r/commit/3584749311801fed0fbf4922a1e1cb749a852112 on my `duet_sim` branch which is based on your `vk-octoprint` branch.
In case you're interested i will make a pull request.

mloidl, 2018-09-22:
Thanks for your feedback.
I'll try to remove the checkboxes and play around with radio-buttons and enums.
Once finished i'll push it to my `duet_sim` branch and i'll create a pull request.

It would be cool when when you consider my changes after the perl-to-cpp rewrite is done.

mloidl, 2020-11-14:
@vojtechkral #5157 should close this bug.
According your proposal i've created a enum to specify which action should be performed after upload.
Instead of using a radionbutton i've used a dropdown because it will not need the dialog to be increased and it looks cleaner.

bubnikv, 2021-11-30:
contributed by @mloidl , merged with 8967ee57ed90f984859d4fbdf9bae78c1683d23d
thanks for contribution.

closed 2021-11-30 by bubnikv