prusa3d/PrusaSlicer #1242: Error in slicing and Gcode producing

Etchy1, 2018-09-19:
### Version
_1.41.0+win64_

### Operating system type + version
Win 10 Enterprise 2015 LTSB

### Behavior
* When slicing an object in Slic3r PE it looks fine in both preview and layers but Gcode only have 13 layers to print. 
* Slice an object in Slic3r PE 
  - Export G-code
  - Open G-code in Repetier Host
  - see that the G-code is not complete.
* _Expected Results_
  - a complete Gcode to print
* _Actual Results_
  - missing layers so faulty result

hope you can fix this, cause i can not get the slic3r and therefore the 3d printer to work

[image]
[image]

  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

FidelCapo, 2018-09-19:
Can you please send me a STL, config files and gcode? Thank you.

Etchy1, 2018-09-19:
Hi Fidel Capo

her are the three files. 

[Prusa.zip](https://github.com/prusa3d/Slic3r/files/2397326/Prusa.zip)

FidelCapo, 2018-09-20:
Thank you for the report. It looks like the gcode is okay and the bug is in the repetier. After upgrading to 2.1.3, the gcode is displayed correctly. Do you have the latest version of repetier?

Etchy1, 2018-09-20:
Hi Fidel
no my repetier is v2.0.5, but the result i see in repetier is also the result i get on the printer itself. 

I have attached an image of the result when printing it on the Prusa. 
[image]

FidelCapo, 2018-09-20:
Did you print through an SD card or through a repetier? Try please upgrade repetier and print again.

Etchy1, 2018-09-20:
I printed via SD. I only opened the file in Repetier host to see if it was fully sliced. The Repetier is for use with a CNC actually, so it was the only software i could find that would display the Gcode as a model.

FidelCapo, 2018-09-20:
I tested your gcode on mac on repetier 1.1.0 and simplify3d. In both cases, the gcode was complete. On Win in the older repetier where gcode was not complete and the latest version of the repetier where gcode was complete. When I open gcode, all looks complete and ok.

Etchy1, 2018-09-20:
hmm ok.. I am trying to print it again now on the Prusa, will let you know if it comes out fully printed or if it's still faulty.. I have reinstalled the Slic3r PC and Prusa Control

FidelCapo, 2018-09-20:
I am trying print this too on MK2MM. Now it is above 13 layers and still printing.

Etchy1, 2018-09-20:
This time it also goes higher for me here.. but had several failed prints that i cant explain then.

Etchy1, 2018-09-20:
it same out complete, so I dont understand why I had so many failed prints

FidelCapo, 2018-09-25:
So I will close this issue. Feel free to reopen it if the issue appears again.

closed 2018-09-25 by FidelCapo