prusa3d/PrusaSlicer #1243: Improve default file name in Slicer PE

gdenicolas, 2018-09-19:
### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

tanguym, 2018-09-23:
This isn't a fix in slic3r, but it should make your life easier:
Since gcode is just text, and the gcode contains how long it will take, you can extract this data from the gcode and put it in the file name through a script. I made a powershell script for myself that does this:

`$lines = Select-String -Path "*_MK3.gcode" -Pattern "estimated printing time \(normal mode\)";
foreach ($line in $lines) {
    $filename = $line.Filename
    $text = $line.Line -replace ' ', '_'
    $time = $text.Substring(41, $text.Length-44)
    $newName = $filename.Substring(0, $filename.Length-10) + $time + ".gcode"
    Rename-Item -Path $filename -NewName $newName
}`

The script will take all files in the folder that end in _MK3.gcode (since that is what my files are named, change that if you have a different printer) and then replace _MK3 in the filename with the printing time. You could also modify the script to add infill etc. to the name.

In case your windows is blocking powershell scripts, you might have to run this first to make it work:
Set-ExecutionPolicy Unrestricted -Scope CurrentUser -Force

bubnikv, 2018-09-27:
This is a duplicate of https://github.com/prusa3d/Slic3r/issues/922
We are working on it.

closed 2018-09-27 by bubnikv