prusa3d/PrusaSlicer #1244: Distorted face selection affecting part boundaries

Maxx84, 2018-09-19:
### Version
1.41.0+win64

### Operating system type + version
Windows 10 Pro (v. 1803)

### Behavior

[Video showing the issue](https://youtu.be/E1yH6OUMrzU)

I created a 2-part assembly in Solidworks 2016 and saved it in stl format, creating two separate stl files (attached). 

I import the first part. Then I double click on the first imported part and add the second one to it.
The "Place object on selected face" feature is now messed up. 
There is now a distortion in the face selection. 
The feature itself works just fine, but the distortion affects the object boundaries.

This does not happen if in Solidworks I save all the components of the assembly in a single stl file.

_Expected Results_
The face selection feature should highlight faces that are on the object itself. I should be able to place the object anywhere on the print bed.

_Actual Results_
The face selection is distorted and the footprint of the object is now wider. This affects the "out of boundary" detection.

#### STL/Config (.ZIP) where problem occurs
[test.zip](https://github.com/prusa3d/Slic3r/files/2399009/test.zip)

Maxx84, 2018-09-19:
**UPDATE**

This bug does not depend on the stl files that I created. 
It happens every time I add an stl, double click on it, click on load part, and add another stl to the assembly.

[New video: adding two times the file PrusaBeerOpener.stl](https://youtu.be/R2FG39ayeV0)

lukasmatena, 2018-09-21:
Thank you for reporting the bug. Did you compile Slic3r yourself? It appears that the bug was introduced after the official release of 1.41.

Anyway, it should be fixed in the master branch with https://github.com/prusa3d/Slic3r/commit/cce5f3203b82f0e2d22ca8c81b91b9e47ea93feb, thanks to @enricoturri.

I suggest leaving the issue open for now, until @FidelCapo confirms that the fix is ok (SPE-520).

Maxx84, 2018-09-22:
Thanks!

Another workaround is to create a single stl of the assembly, if possible. 

After adding the part in Slic3r, it can be split into multiple parts by double clicking on it and selecting Split part.

commit 025fdc13 (2018-10-03): Fixed #1244 - c++ part
commit 3835a1ca (2018-10-10): Fixed #1244 - perl part