prusa3d/PrusaSlicer #1246: Exporting of G-code takes forever

DominikPalo, 2018-09-22:
### Version
1.41.0+

### Operating system type + version
macOS High Sierra 10.13.6

### Behavior
It happens so often that after I slice some more complex STL files, it's not possible to export the G-code. After clicking to the "Export G-code..." button, a message "Exporting G-code" appears in the bottom status bar and stays forever. Also, Slic3r is consuming almost 100% of CPU. I tried to wait for an hour and nothing happens, the only solution is to kill the Slicer process. On the standard Slic3r (1.3.1-dev, not Prusa Edition) it works with the same config without problems and the G-code is exported in few seconds.

#### STL/Config (.ZIP) where problem occurs
[slic3r_report.zip](https://github.com/prusa3d/Slic3r/files/2407983/slic3r_report.zip)

amorphic, 2018-09-24:
I've also seen this in recent Linux releases (I think it started in one of the 1.40 pre-releases). Exporting is so slow that Slic3r is practically unusable. Before this bug appeared Exports were always complete in a matter of seconds but now they are always measured in minutes.

Running `top` shows the export process maxing out a single core like so (note the 100% CPU):

`20246 james     20   0 2299280 1.086g  42376 S 100.0 14.5  18:13.32 perl5.22.0`

bubnikv, 2018-09-24:
>  I think it started in one of the 1.40 pre-releases

Did you compile Slic3r yourself, or is it one of our official builds?

amorphic, 2018-09-24:
@bubnikv always using the official releases. If I get time later today I'll try to go back through the releases I have locally and find out exactly which one introduces the issue. But I *think* it was one of the early 1.40 releases.

macdarren, 2018-09-24:
I too have seen this on my Mac OS 10.13 systems....and I only use official releases as well.... I concur even with my simple models it now takes quite bit longer than the few seconds that it used to take.  My stuff is simple enough that it is not a huge problem to wait the 20-40 seconds but clearly something is different.  Also I agree it seemed to start in the early 1.40 releases though I don't always keep up with the latest version every update.

bubnikv, 2018-09-25:
@FidelCapo Would you please test with the provided data? Thanks.

FidelCapo, 2018-09-25:
It looks like a issue with a "Avoid crossing perimeters" function which is already reported under #287
With the feature disabled, G-code export is fast.
Can you confirm usual G-code export with the feature disabled?

bubnikv, 2018-09-25:
The "avoid crossing perimeters" is mostly useful for leaky extruders,
usually a bowden setup. As our machines are direct drive only, we did not
invest time into improving the "path planning" algorithms for this feature.
We recommend to disable this feature.

I am tempted to close this issue, as it has been reported in #287. Is
anybody of a different opinion? Do you still experience a slow export with
the "avoid crossing perimeters" feature off?

DominikPalo, 2018-09-25:
Thanks - I have more 3D printers at home (e.g. I have Prusa i3 and Ender-3 with a bowden) and I would like to use single Slic3r for both machines.... but now it's not possible. :( And as I know a lot of people use Slic3r PE for bowden machines (because is much better than the original Slic3er).

bubnikv, 2018-09-27:
Thanks for investigation. This is a duplicate of https://github.com/prusa3d/Slic3r/issues/287, closing this one.

closed 2018-09-27 by bubnikv