prusa3d/PrusaSlicer #1247: Slic3r PE Layer Editing Button Greyed Out

Kremund, 2018-09-23:
### Version
1.410+win64

### Operating system type + version
Windows 10 home 1803

### Behavior
The button that says "Layer editing" in slic3r PE is greyed out, even if I select a model. I'm using the Ender 3, but I really don't think that would make a difference. See attached screenshot

* _Steps needed to reproduce the problem_
  -Import an STL
  -Select the model
* _Expected Results_
The row with buttons (Add, Delete, Delete all etc) should light up and be clickable
* _Actual Results_
All the buttons light up, except the one that says `layer editing`

[image]

#### [STL/Config (.ZIP) where problem occurs](https://github.com/prusa3d/Slic3r/files/2408555/Slic3rPEGitIssue.zip)
(I know this probably doesn't make a difference, but I'll just throw the files in here as well)

Kremund, 2018-09-23:
I also tried setting Min and Max layer height, no difference

JasonMeyerDesign, 2018-09-23:
Check the "Enable variable layer height feature" checkbox under the Printer Settings, all the way at the bottom.

[image]

Kremund, 2018-09-23:
I've allready done that... Toggling that option doesn't make a difference :/

Kremund, 2018-09-23:
Huh, the problem solved itself, maybe?
I had allready checked ´Enable variable layer height feature´, but I unchecked and checked it, and now it works!
I'm not sure if toggling it was what fixed the issue, but now it seems to work!
Thank you, @JasonMeyerDesign :)

closed 2018-09-23 by Kremund