prusa3d/PrusaSlicer #1249: SLIC3R PE issue with duplicated items during printing.

jhbinfla, 2018-09-24:
1.41.0 Version

WIN 10Home build 10.0.1734 

SLIC3R PE 1.41.0.   Imported Thingiverse STL of TUSH Spool roller and duplicated to print 2 copies at same time (4 parts each, duplicated to create 8 parts).  split parts and arranged on printbed as it was too large to fit and generated GCODE, saved to SD card and began print.  2 of the parts did not fill in as it should, it generated a grid type infil instead of filled in.  see attached pics  submitted to https://shop.prusa3d.com/forum/hardware-firmware-and-software-help-f64/help-slic3r-pe-question-with-print-t24724.html#p108763.  It was suggested this was a bug and to submit here so I did.  I am not an experienced user/programmer, but will provide any help as needed.  Thank you.

  * _If this is a command-line slicing issue, include the options used_N/A

  * _Screenshots from __*Slic3r*__ preview are preferred_  Do not have any because I did not realize till it was printing. N/A I did not keep the STL, I sliced and generated the GCODE and only saved the GCODE.  I attached the GCODE and the original STL.  All i did was duplicate it in SLIC3R PE, split it, and arrange the parts to fit on the build plate and then generated the GCODE.

_Is this a new feature request?_NO

 where problem occurs  During printing from SD card

[TUSH_SLIC3R PE GCODE issue.zip](https://github.com/prusa3d/Slic3r/files/2412962/TUSH_SLIC3R.PE.GCODE.issue.zip)

FidelCapo, 2018-09-27:
Unfortunately, I am unable to reproduce this issue. Can you please send a 3MF file with this problem?
Plater -> Export plate as 3MF... Thank you

jhbinfla, 2018-09-30:
Just saw your comment, sorry for the delay.

Not sure how to do that.  I do not know how to get the GCODE to be edited.  I did not save the STL file to import into Slic3r PE and cannot seem to be able to view a GCODE once created.  I tried an online program to convert back to an STL, but it did not come out.  Can you help with getting the GCODE into a program to export as 3mf??

bubnikv, 2018-10-01:
To export the 3MF, please go to menu Platter->Export as 3MF.

jhbinfla, 2018-10-03:
FidelCapo and Bubnikv:

Sorry to be dense, but I cannot figure out how to do that.  I tried and nothing happened.  I originally imported the stl, duplicated it and moved the pieces and did not save it as an STL.  ( I am new - sorry) I only exported and saved the GCODE.  It seems that there is no way to get a GCODE into Slic3r ( or any other program that I can tell) and export a 3MF.  Is there?

I am going to try and print the GCODE again tomorrow, but I expect the same thing will happen.  

Is there a way to take a GCODE file and put it in a program to export a 3MF?

FidelCapo, 2018-10-03:
@jhbinfla there isn't way to get STL or 3MF from gcode what I know. Next time you can check gcode before closing Slic3r or deleting objects and export plate as 3MF if will be some errors in gcode again.

jhbinfla, 2018-10-03:
FidelCapo:

Ok.  Yeah, I definately learned my lesson on that one.  Thank you for the reply.  It is a shame because I thought the grid was neat and was hoping to find out how to recreate it for other things if possible.
I will close the incident.

closed 2018-10-03 by jhbinfla

bubnikv, 2018-10-04:
>  It is a shame because I thought the grid was neat and was hoping to find
out how to recreate it for other things if possible.

You can reproduce the grid by setting the number of top / bottom fill
surfaces to zero.