prusa3d/PrusaSlicer #1252: Wrong sliced level surface as "slanted" or little bit rotated

kuldanl, 2018-09-25:
### Version
Slic3r Prusa Edition 1.41.0+win64

### Operating system type + version
Windows 10 1803 OS Build 17134.285

### Behavior
* Object with level surface is sliced as it is little rotated - no sliced level face. Same object with same settings was sliced right in Slic3r 1.40.0.

STL, config and screenshots from both versions of Slic3r are attached.
[mrizka_sudu_dvojita.zip](https://github.com/prusa3d/Slic3r/files/2416251/mrizka_sudu_dvojita.zip)
[image]
[image]

bubnikv, 2018-09-26:
This happens if the last slicing plane crosses the top of your object exactly. When you rotate the model, due to numeric rounding, some of the vertices of the top layer will fall below the cutting plane and some of the vertices will fall above the cutting plane. You will fix your model, if you change the layer height just a tiny bit up or down.

bubnikv, 2018-09-27:
Is this explanation clear to you?

kuldanl, 2018-09-27:
OK, I am understand to this problem. I will recheck gcode with layer height slightly modified, but with other model sliced with layer height in between 0.18-0.21 mm no level surface was sliced. I was surprised with differences in slicing between Slic3r versions.

bubnikv, 2018-09-27:
thanks, closing

closed 2018-09-27 by bubnikv