prusa3d/PrusaSlicer #1253: Appimage build process undocumented

jhoblitt, 2018-09-25:
The appimage build procedure appears to be undocumented/scripted within in this repo.

jhoblitt, 2019-03-26:
ping.  This is a pretty major issue for linux devs that want to contribute.

vojtechkral, 2019-03-27:
I agree it would be nice to have appimage building done as part of the project.
I don't see how it's stopping you from contributing on Linux, can you elaborate?

jhoblitt, 2019-03-27:
As there's no public CI, the build environment for a release needs to replicated as the dev env. Otherwise, there's no way to tell if a contribution will even compile in the release env.  As an inverse example, I am able to build the current master on fedora 29 but it segfaults at start up.  Is it because there's a different version of gcc, different tbb headers, etc.? I don't know and I don't want to have to debug it just for a simple contribution. Additionally, I do slicing on multiple different hosts and `appimage` is rather more convenient then having to build from source.

probonopd, 2019-09-21:
It would be nice if a system such as Travis CI would be used to produce the AppImage. I can help doing this if people would like.

LongLiveCHIEF, 2020-02-09:
The original Slic3r had a [package](https://github.com/slic3r/Slic3r/tree/master/package/linux) directory at the root of the project, and used travis to create the appimage.  I suspect this project just removed that bit after forking, but the build is likely the same or similar.

bubnikv, 2020-02-09:
@LongLiveCHIEF 
> The original Slic3r had a package directory at the root of the project, and used travis to create the appimage. I suspect this project just removed that bit after forking, but the build is likely the same or similar.

Not true at all. PrusaSlicer was forked a lot earlier than that. To some extent we reuse the fruits of labor of each other. For example, the upstream slic3r was at its death bed 4 years ago because nobody was able to produce binary builds for all the pertinent platforms (slic3r-1.2.9 installers were built using Citrus Perl installer, which was long obsolete and it bundled old unstable UI libraries), until Prusa Researched hired and paid me to solve the binary distribution problem, which solution then has been adapted by the upstream slic3r. The draft of the CMake build system was provided by @CRaimer and finished by Prusa Research, then adopted by upstream slic3r. We at PrusaResearch are following the upstream development and we adopted most of their new C++ unit tests for example.

The upstream slic3r guys developed their travis build system long time after we had our internal build system. Our internal build system suits us well because it compiles quicker, which makes our continuous integration turnaround quicker than if using external servers.

LongLiveCHIEF, 2020-02-09:
I'm not accusing you guys of anything, just trying to figure out how the
appimage is built because I want to fix the annoying icon bug.

probonopd, 2020-02-10:
In the meantime, you could extract the contents of the AppImage using `--appimage-extract`, make your changes, and repack the AppImage using https://github.com/AppImage/AppImageKit/releases/download/12/appimagetool-x86_64.AppImage.

hradec, 2024-03-21:
I have added a build script that uses docker container to build prusa slicer easier in any Linux distro and I've sent a pull request for it here: https://github.com/prusa3d/PrusaSlicer/pull/12488

Unfortunately, I couldn't find any documentation or any build files required to create the appimage either. 

It would be really amazing if I could just add the appimage build to this docker script too. 

So +1 on the appimage build documentation, please! :)

wschadow, 2024-03-22:
for the appimage please see here https://github.com/prusa3d/PrusaSlicer/pull/12383

closed 2026-09-29 by prusaslicer-bot[bot]