prusa3d/PrusaSlicer #1255: [Feature request] limit layer height of support material

jhoblitt, 2018-09-25:
The other day I was trying to print a spherical object with openings in the side that result in bridge near the top of the object.  Using supports between the platter and the bottom of the sphere is desirable as even a wide brim still allowed for some "rocking".  However, slic3r would generate support all the way up with the slits in the sides of the sphere to support the bridges at the top which would result in tall supports bridges that don't need it.  A parameter along the lines of max support height by layer # or millimeters would also the supports to be essentially restricted to be just between the platter and the bottom of the sphere.

bubnikv, 2018-09-27:
You may now block the supports with the support blockers
https://github.com/prusa3d/Slic3r/releases/tag/version_1.41.1-beta

Does it work for you?

jhoblitt, 2018-09-27:
I just tried the beta and was able to achieve the effect I was looking for!

Setting used:

Print Settings -> Support Material:
- Generate support material [x] 
- Auto generated supports [ ] 
- Enforce support for the first `50` layers

Is that the feature you were describing?  Looking at 2a81408e8bd64bd3f28319b418ab933895b301d9 (and not understanding this code base at all...) it appears this might be controllable per object?

jhoblitt, 2018-09-27:
Aww, OK, it looks like the default sizing of the Settings window on a 1920x1080 display was causing the `Object Settings` dialog to be completely collapsed -- weird!  Dragging and resizing the window makes the dialog appear.

jhoblitt, 2018-09-27:
And I was able to use a lambda-box as a support blocker. :+1:

bubnikv, 2018-10-05:
So are your needs satisfied now?

jhoblitt, 2018-10-05:
Indeed -- Thank you!

I wasn't sure if I should close the issue since I can't remove the "awaiting" label?

bubnikv, 2018-10-05:
Thanks, closing.

closed 2018-10-05 by bubnikv