prusa3d/PrusaSlicer #1256: stl import failure...

carterm2, 2018-09-26:
### Version
1.41.0+

### Operating system type + version
Mac OSX 10.13.6
### Behavior
Import of STL fails, often large triangles filling interior space where there should not be any triangles
Import stl file... often made in MOI 3d or Rhino 3d 5.5 
Same file imports into Cura and Simplify3D properly
Clean import...  

_Is this a new feature request?_

No but it would be good to get this corrected... otherwise your software can not be used...
[image]

#### STL/Config (.ZIP) where problem occurs
spiralfoot.zip
[spiralfoot.zip](https://github.com/prusa3d/Slic3r/files/2417853/spiralfoot.zip)

FidelCapo, 2018-09-27:
Thanks for the report. Unfortunately, I am able to reproduce this issue. We will try to improve the repair of the models in the future in such a way that we can also repair the very damaged models like this one.

carterm2, 2018-09-27:
Hurka,

This file was not damaged… I created the geometry in Rhino3d and exported the STL file into Cura and Simplify3D without any issues.  I have not been able to read the clean STL file into  SLIC3R.  In fact, this has happened a number of times before, SLICR3 often can not read in STL files without problems while other apps import the files without issues.  I also have tested the STL files in Modo and MeshMixer and MeshLab.  I am convinced that Slicr3 has import problems for mesh files that are not damaged.  Also SLICR3 has a command to “fix” mesh files but this doesn’t fix the file either.  Both SLICR3 and it’s fix file OBJ files have random triangles that are not present in the original file when read into Rhino, Modo, MOI3D, Cura, Simplify3D, MeshMixer or MeshEdit.

Please take a closer look...  I am not stuck on this as I have workarounds and submitted this file and issue out of interest of getting SLICR3 as good as it can be...

The orange image is the STL file correctly displayed.

Here is the SLICR3 import of the same file...

[image]

Regards,

Carter

jkavalik, 2018-09-28:
The file definitely is not ready for printing. The model is not watertight. The blue lines are open edges:

[image]

[image]

And there is the file partially repaired (watertight but may contain some flipped faces or small self-intersections) which works fine for me in Slic3r: [spiralfoot_repaired.zip](https://github.com/prusa3d/Slic3r/files/2427292/spiralfoot_repaired.zip)

BirkBinnard, 2018-09-30:
jkvalik is correct - the geometry has naked edge problems. Here is a bottom-up screenshot from Rhino showing the naked edges in red:
[image]

I use Grasshopper + Rhino for all my design work and I know from experience that Rhino's naked edges need to be avoided like the plague. Basically it's best to always deal with solid, closed Breps, especially when doing Boolean operations - which is how I suspect you made the spiral slot.

bubnikv, 2018-10-03:
The following commit fixes slicing at least for this particular object. There is now no hole filling performed in 3D, but gaps of maximum 2mm are now closed in 2D slices.
https://github.com/prusa3d/Slic3r/commit/69449781cebb470dcd3a9e0f264d707654e4d7ad

We will now test the fix on plenty other objects. Keep your fingers crossed.

bubnikv, 2018-10-05:
This will be fixed in 1.41.1-rc. Closing.

closed 2018-10-05 by bubnikv

carterm2, 2018-10-10:
Thanks guys... I will be testing it going forward... yes I am building with Rhino and using Grasshopper as well at times... I also use Modo, Alias and Solidworks... I have been building products and rapid prototyping my geometry for many years.  I appreciate you guys taking a good link at the problem.  I just wanted you to know I was able to use Cura and other apps to read the STL files from both Rhino and MOI 3d but Slic3r was not able to read the files. 

I  created supports for them in MeshMixer... they turned out to be too bouncy and am using a different design for the feet and getting very clean prints on my MK3 -using IGUS bearings...thinking of buying a second printer...

Thanks...

Carter

commit 517b96ee (2018-10-05): Yet another improvement in closing gaps in slices.
Fixes #1256 and it finally fixes #1119 as well.