prusa3d/PrusaSlicer #1258: No manipulation of support settings on part level possible [1.41.1-beta]

Boby71, 2018-09-27:
### Version
Slic3r 1.41.1-beta - Prusa Edition

_Use `About->About Slic3r` for release versions_

### Operating system type + version
Win10

### Behavior
When double-clicking on a part (which was previously split apart from a two-object stl), I can manipulate more or less all settings for the certain part - but not the support generation:
[image]

I was expecting at least to turn on/off the support generation for the selected object (none, always, only from building plate)

#### STL/Config (.ZIP) where problem occurs
[STL_and_Config.zip](https://github.com/prusa3d/Slic3r/files/2424444/STL_and_Config.zip)

bubnikv, 2018-10-05:
The support generator could only be enabled / disabled per object, not per object part. Is it what you are asking for, to have it configurable per object part? Would you give me a real world use case for that request?

Boby71, 2018-10-06:
Here's a picture of what I meant: I want to print everything WITHOUT support - except the green marked object:
[image]
With that example I get the settings with a double click on the green object, because there are more STL's on one platter. But if two objects are in the same STL and I split them into two seperate objects, I want to be able to adjust support settings for each object as well.

bubnikv, 2018-10-06:
I did exactly that today. Just make sure you set the support attribute at
the object level, not at the object's model level.

bubnikv, 2018-12-12:
I believe it works, and @	Boby71 did not respond. 
Also the upcoming Slic3r 1.42 alpha will make editing of these attributes much more obvious.
Closing.

closed 2018-12-12 by bubnikv