prusa3d/PrusaSlicer #1260: Hole in model leads to missing layers [Slic3r 1.41.1-beta]

Boby71, 2018-09-27:
### Version
Slic3r 1.41.1-beta

### Operating system type + version
Win10-64

### Behavior
Just designed a new model, a simple box with a hole in its wall. Without the hole everything is fine, but as soon as the hole is there, the upper and lower most layer of the hole is missing in the whole model:
[image]

#### STL/Config (.ZIP) where problem occurs
[Hole_Layers_Missing.zip](https://github.com/prusa3d/Slic3r/files/2425492/Hole_Layers_Missing.zip)

Boby71, 2018-09-27:
Just downgraded to 1.41.0 (WIn64) and can confirm, this is a new bug - it's not appearing at the last version:
[image]

MarcoZ76, 2018-09-29:
this behavior happen also with mac os Mojave

Stipena, 2018-10-01:
Having same issue. On exact same part it skips one layer, but sometimes two layers one at bottom one at top of the hole.

MarcoZ76, 2018-10-02:
> this behavior happen also with mac os Mojave.
> Please solve ASAP

This stl is come from original Prusa github repository. Is the new Ensy base.
Version Slic3r PE : 1.41.1 beta
Operation system:  Mac Os 10.14 Mojave

[image]
[image]
[image]

I can confirm that using Slic3R version 1.41.0 it sliced correctly

bubnikv, 2018-10-05:
This is a regression issue, and a duplicate of #1265.
It will be fixed in 1.41.1-rc. Closing.

closed 2018-10-05 by bubnikv