prusa3d/PrusaSlicer #1264: [Feature request] Add comments to gcode for infill, shell, support etc

ogland, 2018-09-28:
I would love to be able to slice my multi material files with Slic3r PE for post processing in Chroma (and possibly other functions). But as of now, it doesn't seem to be possible as far as I know.

Please do implement this, in order for Slic3r PE to keep being super relevant :)

https://support.mosaicmanufacturing.com/hc/en-us/articles/115002756094-Minimizing-Filament-Waste

bubnikv, 2018-09-29:
There is a switch in Slic3r to enable verbose g-code output. Does it
fulfill your needs?

ogland, 2018-09-29:
It doesn't seem like it worked for mosaic well enough for them to use it.

They write this about purging to infill using Chroma:
"This feature works with Simplify3D, Cura, and KISSlicer PRO. Slic3r's G-code comments do not yet distinguish between infill and shells."

Read more about it by following this link: https://support.mosaicmanufacturing.com/hc/en-us/articles/115002756094-Minimizing-Filament-Waste

ogland, 2018-10-01:
It seems like Slic3r PE verbose gcode only tags infill as "fill", and it also tags top infill, bottom infill etc as "fill". I think this could be easily fixed.

Here is another issue from 2014 about this:
https://github.com/slic3r/Slic3r/issues/2079

cfelicio, 2018-10-15:
I reached out to Mosaic as well to see their take on this, and looks like they will get this updated on Chroma side as soon as it is implemented :)

[image]

mylife4aiurr, 2018-12-09:
Yes please add this feature...... Needed

bubnikv, 2019-05-23:
Martin Temmink commented on Facebook:
(PrusaSlicer 2.0.0) Works great with Palette 2 pro and the post-processor p2pp!

So could we close this issue?

ogland, 2019-05-23:
If it works with the original palette, sure!

bubnikv, 2019-05-24:
so would you close this issue then? my understanding is, that it is not
needed anymore.

cfelicio, 2019-05-24:
@bubnikv It does not work yet. Can we make the verbose cover all feature types or at least main feature types (see #2289 ) instead of Palette specific?

ogland, 2019-05-24:
It doesn't work with the original palette so issue should remain open.

bubnikv, 2019-05-24:
@cfelicio thanks for the hint. Let's close this one and keep #2289 open.

closed 2019-05-24 by bubnikv