prusa3d/PrusaSlicer #1267: [ Feature Request ] Allow customization of settings presented on the Plater tab.

jmckeown2, 2018-09-29:
### Version
1.41.0+

### Operating system type + version
MacOS 10.13.6 (*High Sierra*)

### Behavior
* Platter tab in GUI exposes common settings conveniently e.g. Fill Density, Support & Brim
* _Steps needed to reproduce the problem_
  * NONE
* _Expected Results_
  * I Would like to be able to customize which settings appear on the Platter tab. So Maybe I want to be able to edit not just the infill % but also the pattern, or maybe the number of Perimeters, or the support angle or Seam Location. As I user, I would like to be able to control which Settings are exposed to the Platter page so that I can most conveniently adjust the settings that matter to me. 
* _Actual Results_
  * Have to Jump from tab-to-tab anyway, limiting usefulness of convenient settings controls 

Is this a new feature request?
**YES**

#### STL/Config (.ZIP) where problem occurs
N/A

faboaic, 2018-09-30:
+1 for Windows 64bit 
I often wished there was that feature

bubnikv, 2018-12-12:
The upcoming Slic3r 1.42 alpha will allow you to add these values to the object list per object. Please let us know whether this is sufficient to you.

bubnikv, 2019-05-21:
Let's decrease the number of open issues. We have implemented per object parameters at the plater, and we have added support and infill density to the plater as well. We currently do not support adding of other global parameters to the plater, there is https://github.com/prusa3d/PrusaSlicer/issues/1267 open to discuss that.

haarp, 2019-05-24:
@bubnikv I think you accidently closed #1267 while meaning to close #1497

RyanEwen, 2022-10-19:
I am dying for this. I'm very tired of jumping around different tabs and hunting for the 3-4 settings I often tweak and can't easily handle using saved profiles.

lukasmatena, 2026-09-01:
Implemented in [3.0.0-alpha11](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_3.0.0-alpha11). Closing.

closed 2026-09-01 by lukasmatena