prusa3d/PrusaSlicer #1269: Top/bottom fillpattern not used on all visisble (top) layers

mdbijl, 2018-09-29:
### Version
1.41.0+

### Operating system type + version
MacOS Mojave

### Behavior
* When having surfaces on different heights, only the topmost layer is printed with the **Top/bottom fillpattern**
* Lower top layers are printed with a diagonal pattern
* For example, the top case in this example: https://www.thingiverse.com/thing:2593692

When sliced with MakerWare and printed on Makerbot Replicator 2 (blue):
[image]
Note the horizontal lines throughout the whole hole top of the case

When sliced with slic3r and printed on Prusa i3 MK3 (red):
[image]
Note the **diagonal** lines on the whole top of the case

### Probable cause
* Top layer detection is wrong
* The top layer is not the topmost layer, there can be multiple top layers:
** Every part of a layer that doesn't have any layer above it, should also be considered a top layer

#### STL/Config (.ZIP) where problem occurs
[maker-tutorials.com-mechanical-gamepad-top.stl.zip](https://github.com/prusa3d/Slic3r/files/2430981/maker-tutorials.com-mechanical-gamepad-top.stl.zip)

bubnikv, 2018-10-01:
I am not sure I understand your issue. The infill pattern seem to me just
fine. By default, Slic3r hatches infills in a diagonal direction.

Would you please provide your config file, or generate a 3mf file?

bubnikv, 2018-12-12:
No response, closing.

closed 2018-12-12 by bubnikv