prusa3d/PrusaSlicer #1271: Slic3r (all versions) not running under Win 10 Home

BadRaven, 2018-09-30:
### Version
Slic3r Prusa Edition, Slic3r (various versions) 

### Operating system type + version
Windows 10 Home, 1803, 24/5/18, Build 17134.285

### Behavior
Was working from PrusaControl, Pronterface and Desktop Icon

Updated Prusa set of files via Prusa support page including Slic3r 1.41.0 Prusa Edition

PrusaControl still operates, slices usable files, but Slic3r will not run from the desktop icon, nothing happens when icon double clicked. Pronterface runs until a .stl is loaded, when it reports missing file and stops loading

This is a consistent repeatable problem.

Uninstalled and reinstalled whole package, same problem.
Uninstalled and reinstalled Slic3r, same problem. (and again also taking out things uninstall left like directories)
Uninstalled and installed version from Slic3r sources, same problem.
Uninstalled just Slic3r, reinstalled, repeated with several versions, NONE will run.

Prusa Control operates normally throughout with any version.

Attempting to run Slic3r from its desktop icon does nothing at all.

Pronterface runs, but when a .stl file is loaded it errors like this example:

Slicing D:\A_Prusa_Files\A_3D_Print\Quad Parts\Feather_V1.stl
Slicing Slic3r/slic3r.exe D:\A_Prusa_Files\A_3D_Print\Quad Parts\Feather_V1.stl --output D:\A_Prusa_Files\A_3D_Print\Quad Parts\Feather_V1_export.gcode
[ERROR] Failed to execute slicing software: 
Traceback (most recent call last):
  File "printrun\pronterface.pyc", line 1577, in skein_func
  File "subprocess.pyc", line 711, in __init__
  File "subprocess.pyc", line 948, in _execute_child
WindowsError: [Error 2] The system cannot find the file specified

Slic3r Config file is attached.
[Slic3r not running1.zip](https://github.com/prusa3d/Slic3r/files/2431301/Slic3r.not.running1.zip)

ALL the file sets run OK on a Win 7 laptop, ALL (now) give the same issue on Win 10 laptop.  Prusa unable to help further.

I assume Prusa Control runs Slic3r PE to slice files, which works, but Slic3r will not run in any other way.

bubnikv, 2018-10-01:
It is hard to say what is wrong with your setup.

What will happen, if you unpack the win64 zip file from
https://github.com/prusa3d/Slic3r/releases
onto desktop and run Slic3r from there?

If it does not run from there, please run slic3r-console.exe from command
line and post the output of slic3r-console.exe here. Thank you.

BadRaven, 2018-10-01:
OK, Unpacked to desktop.
When Slic3rf run directly no action and no text in cmd window.

C:\Users\Bad Raven\Desktop\Slic3rPE-1.41.1-beta+win64-full-201809261048>slic3r.exe

then nothing........................  When Slic3r run from console in cmd window I get the following:-

C:\Users\Bad Raven\Desktop\Slic3rPE-1.41.1-beta+win64-full-201809261048>slic3r-console
Slic3r installed in a loclized path. Using an 8.3 path: "C:\Users\BADRAV~1\Desktop\SLIC3R~1.1-B\"
Can't locate warnings/register.pm in @INC (you may need to install the warnings::register module) (@INC contains: .) at vars.pm line 7.
BEGIN failed--compilation aborted at vars.pm line 7.
Compilation failed in require at Cwd.pm line 4.
BEGIN failed--compilation aborted at Cwd.pm line 4.
Compilation failed in require at FindBin.pm line 83.
BEGIN failed--compilation aborted at FindBin.pm line 83.
Compilation failed in require at C:\Users\BADRAV~1\Desktop\SLIC3R~1.1-B\slic3r.pl line 7.
BEGIN failed--compilation aborted at C:\Users\BADRAV~1\Desktop\SLIC3R~1.1-B\slic3r.pl line 7.

C:\Users\Bad Raven\Desktop\Slic3rPE-1.41.1-beta+win64-full-201809261048>

NameTakenWasTaken, 2018-10-02:
I have exactly the same problem. When trying from the command line, I get:

C:\Prusa\Slic3rPE-1.41.1-beta+win64-full-201809261048\Slic3rPE-1.41.1-beta+win64-full-201809261048>slic3r-console.exe
Loading an application config file failed:
<unspecified file>(1): '=' character not found in line

bubnikv, 2018-10-03:
Will slic3r start, if you execute
perl5.24.0.exe slic3r.pl 
as discussed in https://github.com/prusa3d/Slic3r/issues/1285?

bubnikv, 2018-10-03:
> No, I get.....................
>
> Microsoft Windows [Version 10.0.17134.285]
> (c) 2018 Microsoft Corporation. All rights reserved.
>
> C:\Users\Bad Raven>cd\
>
> C:>perl5.24.0.exe slic3r.pl
> 'perl5.24.0.exe' is not recognized as an internal or external command,
> operable program or batch file.
>
> C:>

You have to run perl5.24.0.exe from the installation directory of slic3r.

BadRaven, 2018-10-03:
C:\Users\Bad Raven\Desktop\Slic3r>perl5.24.0.exe slic3r.pl
Can't locate warnings/register.pm in @INC (you may need to install the warnings::register module) (@INC contains: .) at vars.pm line 7.
BEGIN failed--compilation aborted at vars.pm line 7.
Compilation failed in require at Cwd.pm line 4.
BEGIN failed--compilation aborted at Cwd.pm line 4.
Compilation failed in require at FindBin.pm line 83.
BEGIN failed--compilation aborted at FindBin.pm line 83.
Compilation failed in require at slic3r.pl line 7.
BEGIN failed--compilation aborted at slic3r.pl line 7.

C:\Users\Bad Raven\Desktop\Slic3r>

bubnikv, 2018-10-03:
Do you have the following file on your hard drive?
C:\Users\Bad Raven\Desktop\Slic3r\lib\warnings\register.pm

BadRaven, 2018-10-03:
Yes

bubnikv, 2018-10-03:
I am quite clueless. Isn't there any issues with the access rights to these
files? Are they read/write enabled to your current Windows user?

BadRaven, 2018-10-03:
Can I take a different tack?

Is or How is Prusa Control accessing Slic3r PE?  (As this works OK)

bubnikv, 2018-10-03:
> This is the whole register.pm file:-
>
> package warnings::register;
>
> our $VERSION = '1.04';
> require warnings;
> left here as cruft in case other users were using this undocumented routine --
> rjbs, 2010-09-08
>
> sub mkMask
> {
> my ($bit) = @_;
> my $mask = "";
>
> vec($mask, $bit, 1) = 1;
> return $mask;
>
> }
>
> sub import
> {
> shift;
> my @categories = @_;
>
> my $package = (caller(0))[0];
> warnings::register_categories($package);
>
> warnings::register_categories($package . "::$_") for @categories;
>
> }
> 1;
> *END*
>
> =pod
>
> =head1 NAME
>
> warnings::register - warnings import function
>
> =head1 SYNOPSIS
>
> use warnings::register;
>
> =head1 DESCRIPTION
>
> Creates a warnings category with the same name as the current package.
>
> See L for more information on this module's usage.
>
> =cut
>
> it has no special access requirements!

> Is or How is Prusa Control accessing Slic3r PE? (As this works OK)

Prusa Control contains some old version of Slic3r PE. There should be no
difference between how Prusa Control launches Slic3r and how Slic3r
launches itself.

bubnikv, 2018-10-03:
@NameTakenWasTaken Do you have this issue just with 1.41.1-beta, or do you have it with 1.41.0 final release as well?

NameTakenWasTaken, 2018-10-03:
> @NameTakenWasTaken Do you have this issue just with 1.41.1-beta, or do you have it with 1.41.0 final release as well?

I see the problem with all versions tried, including 1.41.00 and 1.40.0 rc2.

bubnikv, 2018-10-05:
Will it help to run slic3r with the --datadir parameter to direct Slic3r to use an alternate user configuration directory? Will something like the following line work?

perl5.24.0.exe slic3r.pl --gui --datadir %temp%\datadir

BadRaven, 2018-10-05:
YES!!!!  That runs it coming up with its usual start and an initial configuration wizard window

bubnikv, 2018-10-06:
Would you please zip the content of your slic3rpe directory containing your
user profiles and attach it here for our investigation? Thanks.

BadRaven, 2018-10-06:
Sorry, you need to be more specific as to directory name and files names please.

NameTakenWasTaken, 2018-10-07:
With the help from Prusa, I solved my problem. Fairly simple, actually: I deleted my appdata\roaming\Slic3rPE folder, re-installed, and it worked!

vojtechkral, 2018-10-10:
@NameTakenWasTaken @BadRaven Did you guys both delete the `appdata\roaming\Slic3rPE` directory, meaning none of you have it anymore? Or do you have it backed up somewhere such that you could share it with us for investigation of the problem?

BadRaven, 2018-10-11:
Sorry, Deleted.

bubnikv, 2018-10-11:
Anybody could share his broken content of the Slic3rPE folder?
Otherwise we will close this issue.

bubnikv, 2018-12-12:
No response, closing.

closed 2018-12-12 by bubnikv